ckb-next/ckb-next #390: ckb-next makes shutting down and rebooting slow

predme, 2019-03-22:
# Bug report

## General information

### Source
  * ckb-next-git package in AUR

## Logs & crash reports

[ckb-next-report-1553251526.gz](https://github.com/ckb-next/ckb-next/files/2996149/ckb-next-report-1553251526.gz)
[journalctl.txt](https://github.com/ckb-next/ckb-next/files/2996183/journalctl.txt)

## What is the program's current behavior?
Working as expected other than the slow reboot / shutdown times.

## What is the expected behavior of the program?
To not delay shutdown / reboot.

## Any other notes & comments?
I only added the output that had something to do with input and ckb-next from journalctl in the uploaded log. If that was a bad idea and something crucial is missing, tell me and I'll upload a more complete log. I also noticed that there was an issue similar to mine from the past ([#168](https://github.com/ckb-next/ckb-next/issues/168)) but from what I gathered, it was fixed. Though it seems like said fix doesn't apply for my system since well, it happens to me.

ZirconiumX, 2019-03-22:
I've noticed a few things, so I'll go through these one at a time.

First, you don't have any usbcore quirks on your command line; could you [add them](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks) and see if it makes a difference?

predme, 2019-03-22:
I added the usbcore quirks and it's still the same. The only difference I noticed is that there's now a delay of about 10-15 seconds until the mouse and keyboard becomes responsive after booting.

ZirconiumX, 2019-03-22:
That's odd; @tatokis we've never had the usbcore quirks make things worse,
have we?

I think you need to join us in IRC by clicking the blue Freenode button in
the readme.

tatokis, 2019-03-22:
If you're using USB2 try using USB3, or the opposite. This sounds like a kernel issue, and the usbcore quirks should take care of it. There's nothing in the logs indicating that it's on our end.

predme, 2019-03-22:
Using ZirconiumX's patch ([drop-k95-rgb](https://github.com/ckb-next/ckb-next/tree/drop-k95-rgb)) the issue is worked around for now. The USBCore quirks thing didn't work for me since I accidentally got one too many (and unsupported) USB device included. 

I said it in the IRC already, but thanks again :)

closed 2019-03-22 by predme

OvermindDL1, 2019-03-22:
Might be worth leaving this open until it is fixed in master?  :-)

ZirconiumX, 2019-03-22:
Pretty sure this is the wrong master for it to be fixed in; try
torvalds/linux master :P