ckb-next/ckb-next #393: Firmware thru LVFS / fwupd ?

maage, 2019-03-31:
# Something else

ckb-next does firmware update. But Linux has LVFS / fwupd to do that too.

https://fwupd.org/
https://twitter.com/hughsient/status/1112256495981334528
https://www.phoronix.com/scan.php?page=news_item&px=LVFS-500k-In-A-Month

I wonder if it would be possible to move firmware management and installation to fwupd from ckb-next. Or is this something where Corsair needs to do things.

ZirconiumX, 2019-03-31:
@hughsie actually approached us for that very reason (we're why Corsair are listed there). In a nutshell, we said no because Corsair are prone to making incompatible changes without giving a damn about us, and so we can't guarantee that an LVFS firmware update won't break ckb-next. If that happens, a lot of people are going to be angry at us, so we prefer handling firmware rollout ourselves.

Of course, if Corsair ever change their mind about helping us out, that situation might change.

closed 2019-03-31 by ZirconiumX

maage, 2019-03-31:
Okay fair enough. Thank you!

ZirconiumX, 2019-03-31:
Thinking about it, there are another few points against using the LVFS.

We would still need to support Mac OS, and I'd imagine the LVFS would want us to maintain the firmware update code for Linux, so it doubles the maintenance effort for that section of code for little gain.

Additionally, on both platforms, to update firmware you need to take full control of the device, so fwupd would have to kick us off while this is happening, resulting in the user having no input or RGB while a firmware update is occurring. This could get frustrating, especially if an update takes a while.

hughsie, 2019-04-01:
> I'd imagine the LVFS would want us to maintain the firmware update code for Linux

No, not really; I'm happy to help get the code into fwupd and maintain it there.

> so fwupd would have to kick us off while this is happening, resulting in the user having no input or RGB while a firmware update is occurring

That's pretty normal; we have UI for that in gnome-software. e.g. the Logitech and Wacom plugins both disable the input device during firmware update.