ckb-next/ckb-next #394: [Bug] [Linux] Corsair K68 Left Ctrl & Shift are the same

Cerberooo, 2019-04-02:
# Bug report

## General information

Source: ckb-next (0.4.0-1) package in AUR 

## What is the program's current behavior?

When I start the ckb-next-daemon, the left ctrl and left shift keys behave the same.

Binding tab: 

[image]

When I change the modifier to left-ctrl this happens:

Binding tab for left-ctrl: 

[image]

[image]

[ckb-next-report-1554209745.zip](https://github.com/ckb-next/ckb-next/files/3034033/ckb-next-report-1554209745.zip)

ZirconiumX, 2019-04-02:
cc @tatokis

tatokis, 2019-04-03:
Please check that you don't have Ctrl bound to Shift in the settings tab.
[image]

Cerberooo, 2019-04-03:
Solved it... thanks!
But why this was the default after a fresh install?

closed 2019-04-03 by Cerberooo

tatokis, 2019-04-03:
I'm pretty sure it's not.

What probably happened here is that you accidentally scrolled while moving the cursor over the dropdowns and that changed the setting without you realising.

Cerberooo, 2019-04-03:
You're probably right. Thanks for your help!