ckb-next/ckb-next #395: [Scimitar Pro] keyboard + sidebuttons causes freeze #bug-report

filosofis, 2019-04-03:
- [Bug Report](#bug-report).
Scimitar Pro RGB

# Bug report
Using sidebuttons (1-12)  and keyboard at the same time causes graphics to freeze (audio not affected)

### Source
  * built manually using quickinstall script

## Logs & crash reports
[ckb-next-report-1554301853.gz](https://github.com/ckb-next/ckb-next/files/3039168/ckb-next-report-1554301853.gz)

tatokis, 2019-04-05:
Someone else reported the same issue, and it seemed to be an issue with GNOME/GTK. I don't think this one is on us

Ahajha, 2019-04-26:
I'm having the same issue. When playing Warframe with Lutris, using my keyboard and non-side mouse buttons is fine, but using any side buttons causes stuttering.

Klmakke, 2020-03-11:
I'm also having this issue when running a specific game, Tibia. getting quite the stutter when using the 1-12 keys. Any fix for this? Not run GNOME?

tatokis, 2020-03-12:
If it happens on a specific game, it might be worth checking jstest-gtk and sdl2-jstest to see if the mouse is recognised as a joystick (with ckb running).
We have udev rules in place to make the mouse not be recognised as such a device, but it's possible something else causes it.

Perhaps also check /var/log/Xorg.0.log after unplugging and replugging the mouse to see if there's anything suspicious there.

It might also be worth trying to switch to evdev instead of libinput to see if the issue persists.

Other than that, as I mentioned above, I don't think there's much, if anything, we can do.

filosofis, 2020-03-21:
> I'm also having this issue when running a specific game, Tibia. getting quite the stutter when using the 1-12 keys. Any fix for this? Not run GNOME?
> 
> edi: I tried xfce DE for a sec and it's working a lot better, time to drop GNOME i guess.

I also switched to xfce for this very reason a couple of months ago. And I'm quite pleased that I did. I actually prefer it now.

koplo199, 2021-06-08:
This is definitely a GNOME bug (Xorg only, Wayland not affected), the issue is reported on the gnome tracker and still open, however as a workaround one may compile libmutter with a patch applied, instruction here : [https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/1858#note_818548](https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/1858#note_818548). I believe #662 is a duplicate of this bug.

closed 2022-10-03 by tatokis