ckb-next/ckb-next #396: K65 media keys not working

EvanVujcec, 2019-04-07:
# Bug report

## General information
Media keys on K65 are not functioning. The Mute and Volume Up/Down are mapped correctly in the software but nothing happens when I press them. Can't tell if the Fn+F9/F10/F11/F12 media keys are mapped correctly but they aren't functioning either. Keyboard isn't in BIOS mode. The keys work fine when I boot into Windows, but oddly enough, `showkey` doesn't output anything when I press any of the media keys.

Also don't remember if they were working before the most recent update since I haved booted into this OS in a while.

### Source
Built manually through quickinstall

## Logs & crash reports

[ckb-next-report-1554649126.gz](https://github.com/ckb-next/ckb-next/files/3051639/ckb-next-report-1554649126.gz)

tatokis, 2019-04-09:
Stop the daemon first
```
sudo systemctl stop ckb-next-daemon
```
then run this command:
```
sudo usbhid-dump -e all -t 10000 -d 1b1c:1b13
```

Press and release all the media keys that weren't working, from left to right, top to bottom, then once done, wait for 10 seconds for the program to time out and exit. Do not press any other keys while the command is running.

Once done, please paste the output here.

BloodyIron, 2019-04-14:
Yeah getting same issue here, got K65 regular. I'm on ckb-next 4.0

NeuroticTumbleweed, 2019-04-25:
Hey @tatokis I'm having the same issue. 
I'm running fedora 29 with this as my ckb-next version:
```
ckb-next.x86_64                      0.4.0-1.fc29
```
Here's my output of that usbhid-dump
```
[andrew@MSI-Desktop ~]$ sudo usbhid-dump -e all -t 10000 -d 1b1c:1b07
001:003:002:DESCRIPTOR         1556215244.995238
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 67 15 00 25 01 75 01
 95 68 81 02 09 85 19 87 29 8B 09 90 09 91 15 00
 25 01 75 01 95 08 81 02 C0

001:003:001:DESCRIPTOR         1556215244.995560
 05 0C 09 01 A1 01 19 00 2A FF 0F 15 00 26 FF 0F
 75 10 95 01 81 00 C0

001:003:000:DESCRIPTOR         1556215244.996205
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 05 0C 09 00 15 80 25 7F 75 08 95 08 B1 02 C0

Starting dumping interrupt transfer stream
with 10 seconds timeout.

001:003:001:STREAM             1556215248.321154
 E2 00

001:003:001:STREAM             1556215248.425156
 00 00

001:003:001:STREAM             1556215249.377196
 EA 00

001:003:001:STREAM             1556215249.473200
 00 00

001:003:001:STREAM             1556215250.753208
 E9 00

001:003:001:STREAM             1556215250.857218
 00 00

001:003:002:Interrupt transfer timed out
001:003:000:Interrupt transfer timed out
001:003:001:Interrupt transfer timed out
No more interfaces to dump
```
Please let me know if you need anymore information :slightly_smiling_face:

jonnyeom, 2019-05-03:
I have been seeing this for a month or two as well.
Vengeance K65 User. I have the RGB as well and that works fine.

Here is my dump of `usbhid-dump`
```
001:004:002:DESCRIPTOR         1556893662.027755
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 67 15 00 25 01 75 01
 95 68 81 02 09 85 19 87 29 8B 09 90 09 91 15 00
 25 01 75 01 95 08 81 02 C0

001:004:001:DESCRIPTOR         1556893662.028327
 05 0C 09 01 A1 01 19 00 2A FF 0F 15 00 26 FF 0F
 75 10 95 01 81 00 C0

001:004:000:DESCRIPTOR         1556893662.028954
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 95 01 75 08 81 01 95 05 75 01
 05 08 19 01 29 05 91 02 95 01 75 03 91 01 95 06
 75 08 15 00 26 FF 00 05 07 19 00 2A FF 00 81 00
 05 0C 09 00 15 80 25 7F 75 08 95 08 B1 02 C0

Starting dumping interrupt transfer stream
with 10 seconds timeout.

001:004:001:STREAM             1556893664.598520
 E9 00

001:004:001:STREAM             1556893664.734497
 00 00

001:004:001:STREAM             1556893666.006516
 EA 00

001:004:001:STREAM             1556893666.134511
 00 00

001:004:001:STREAM             1556893668.054535
 E2 00

001:004:001:STREAM             1556893668.166539
 00 00
```

tatokis, 2019-05-05:
Should be fixed in the k65-media-keys branch. Can someone please test it?
https://github.com/ckb-next/ckb-next/tree/k65-media-keys

NeuroticTumbleweed, 2019-05-05:
Thanks I'll try test it out.

NeuroticTumbleweed, 2019-05-05:
It's a success :birthday: 
Steps I took:
1. Removed the fedora packaged version of ckb-next
1. Cloned the repo and switched to branch `k65-media-keys`
1. Installed with the `./quickinstall` script
1. Tested volume and mute buttons (First just after the installation worked and then I also disconnected and reconnected my keyboard) All of the buttons worked.
1. Checked that `ckb-next-daemon.service` showed that it was enabled with `systemctl list-unit-files`

jonnyeom, 2019-05-06:
It s a success for me as well!

tatokis, 2019-05-07:
Merged in master https://github.com/ckb-next/ckb-next/commit/fd509f5bb25aabbcc982bb28cb6f445a875648a0

closed 2019-05-07 by tatokis