ckb-next/ckb-next #397: Corsair Vengeance RGB Support

s7relok, 2019-04-11:
Hello,

Nice project here.

Do you plan to support the corsair vengeance RGB RAM sticks? What can I do for help?

Regards,

airbat, 2019-04-21:
+1 need it to. If i can help in anyway, tell me.

arrowgamer, 2019-05-18:
I'd like this, along with support for all of the devices in the corsair SDK.

ZirconiumX, 2019-05-18:
Support for everything in the SDK is impractical on so many levels, because
the entire development team is two burned out university students who don't
get rewarded for this in any way.

arrowgamer, 2019-05-19:
This sounds like a job for the magic of open source! A few burnt out
students will need contributors.

I'll probably create my own utility to manage my setup, but I'll be doing
it in node because that's a project I can complete in a reasonable amount
of time. I'll be making a cli, so perhaps one day it can be utilized here
for the rest of the world who wants it 🤷‍♂️

airbat, 2019-05-19:
Maybe you can make your own but contribute here or give the sources ?

arrowgamer, 2019-05-19:
I thought about contributing here, really I did, but I'm just not confident
enough in C++, meaning by the time I'd have this repository figured out,
I'd already have been able to make what I wanted for myself.

The source for whatever I make will be available on here or bitbucket, and
I'd always be willing to help out anyone wanting to augment their project
with it.

ZirconiumX, 2019-05-19:
The magic of open source has kinda faded for me.

This is a project that relies heavily on its community to make any progress
at all; nearly all devices supported by CKB needed packet traces taken by
members of the community, and I am grateful for each and every one of you.

But it kinda drags when everybody just disappears over time.

CalcProgrammer1, 2019-06-03:
I've figured out the protocol for the non-Pro sticks and have some code to control them.  I'm reverse engineering other RGB RAM along with Asus Aura so I decided to get my hands on Corsair's RGB RAM as well.  I also got the Pro enhancement kit (the fake RAM modules that just have lights) and am reverse engineering that protocol too.  The project is on GitLab:

https://gitlab.com/CalcProgrammer1/OpenAuraSDK

The Corsair support isn't built into the UI yet.  I built the UI with Aura-specific modes and so I'll need to modify it to work with Corsair.

airbat, 2019-06-04:
If i Can help on anyway to add support for different corsair support, i have the ram, keyboard (Corsair Strafe RGB MK.2), mousse (Corsair Dark Core RGB SE), watercooling AIO (Corsair Hydro Series H115i PLATINUM) and the case (Corsair Crystal 280X RGB with lighting node pro).
I also use windows sometimes si i can use it to help you if needed.
Just tell me what you need and i will do if it's possible.

CalcProgrammer1, 2019-07-02:
Did some work this weekend reverse engineering the Pro RGB modules.  I created a tool to log i2c messages being sent from the iCue software and then captured logs for each mode change.  I have a working driver I haven't committed yet (will push later this week once I get timing properly fixed) that can change modes and set colors.  Details here:

https://gitlab.com/CalcProgrammer1/OpenAuraSDK/issues/31

tatokis, 2019-07-02:
Great, now will you be helping us adding support for it in ckb?

CalcProgrammer1, 2019-07-02:
I'm ultimately working on my own project (trying to unify all the different vendors' RGB stuff into one API) but would be glad to offer assistance adding it to ckb.  I've never messed with ckb's source code and AFAIK it doesn't currently support any SMBus devices, but I can help add this if needed.

olielvewen, 2019-10-24:
@airbat 

> If i Can help on anyway to add support for different corsair support, i have the ram, keyboard (Corsair Strafe RGB MK.2), mousse (Corsair Dark Core RGB SE), watercooling AIO (Corsair Hydro Series H115i PLATINUM) and the case (Corsair Crystal 280X RGB with lighting node pro).
I also use windows sometimes si i can use it to help you if needed.
Just tell me what you need and i will do if it's possible.

Hi,
For your keyboard and your mouse (I've got the same and I'll test soon the wireless branch) I don't know.

But for your watercooling AIO and perhaps for your Corsair Crystal 280 X RGB with lighting node pro, you can use/test the following softwares : 

- [OpenCorsairLink](https://github.com/audiohacked/OpenCorsairLink)
- [LiquidCtl](https://github.com/jonasmalacofilho/liquidctl)
- [Colorsair](https://github.com/Chlorek/Colorsair)
- [Lightning node pro](https://github.com/benburkhart1/lighting-node-pro)

---

Don't think to get in one software all your Corsair products like iCue.

I don't know on which distribution you are but if you are on Manjaro or any Archlinux derivated distribution, it is easily to install and test it. 

Another one word about this projects : 
*OpenCorsairLink* seems to be not very active for issues
*LiquidCtl* is very active for issues and PR if we help us. Recently, all features for PSU RMi and HIX series have implanted (like iCue). The next version 1.3.0 is planned for the 17 November.
*Colorsair* seems not to be active now
*Lightning node pro* has done a PR for Colorsair. To see and to test.

---

LIke you are French, I've a serie of blog entries about Hardware Gaming on Linux. Watch [here](https://linuxevolution.wordpress.com/2018/11/23/me-gamer-and-linux-user-impossible-mission-not-exactly-part-2/)
2 autres articles sont en préparations dont un sur le superbe logiciel de CalcProgrammer1. Ca c'est du lourd. Il est le premier et le seul à avoir compris comment se servir de SMBus. :+1:  Et j'ai même pas eu le temps de tester. Snif 
Un 9eme sera un récapitulatif de tous les autres..... Quand j'aurai le temps.

CalcProgrammer1, 2019-10-24:
I've documented the protocol for the Corsair Pro RAM here:

https://gitlab.com/CalcProgrammer1/OpenAuraSDK/wikis/Corsair-Vengeance-Pro-RGB

gravityfargo, 2024-05-01:
Closing in favor of #589

closed 2024-05-01 by gravityfargo