ckb-next/ckb-next #398: Lag, infinitely repeating characters, missing characters

coma-toast, 2019-04-11:
# Bug report

## General information

### Source
It's been a while since I originally installed it, but I believe I installed via git clone, ./quickinstall. I do get package updates via apt

## Logs & crash reports
[ckb-next-report-1553256508.gz](https://github.com/ckb-next/ckb-next/files/3070276/ckb-next-report-1553256508.gz)
[image]

## What is the program's current behavior?
When running, keyboard input is severely impacted. There is a slight lag, but the pressed keys are sometimes repeated forever, come in out of order, or are missed completely. See attached gif of behavior. It seems like this may be a polling rate issue of some sort. The "save to hardware" button also is greyed out, so I cannot use any lighting features. 

## What is the expected behavior of the program?
Typing to work as normal with the app running. 

## Any other notes & comments?
I did have this working on this same machine in the past, but that may have been on 16.04 (currently on 18.04). 
This is a Corsair Strafe keyboard.

ZirconiumX, 2019-04-11:
You need to add [usbhid quirks](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbhid-quirks) and [usbcore quirks](https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks) to your kernel command line. This *should* fix the issue, but do report back.

coma-toast, 2019-04-13:
I added both quirks, same behavior. I added them as one time boot parameters.

ZirconiumX, 2019-04-13:
They need to be added to GRUB or whatever bootloader you use; they're not "use once and fixed" parameters, but workarounds for issues in Corsair firmware (of which there are many).

It's possible you have a new issue with said firmware, but first could you update your kernel? Newer versions have more workarounds, so the quirks might be ignored in your kernel version.

coma-toast, 2019-04-15:
Kernel updated from 4.15 to 5.07. Same issue.
Added USBHID quirks via /etc/default/grub, update-grub, reboot. Same issue.
Added USBCore quirks via /etc/default/grub, update-grub, reboot. Same issue.

I believe in the past I had the same issue, but there was a `save to device` option that allowed me to close the app. Now, when closing the app, all lights on the kb are off - which is annoying when I come in to the office early and it's still dark.

tatokis, 2019-04-15:
Try plugging the keyboard on a USB2 port if it's on USB3 one, or the opposite. See if that helps.

ZirconiumX, 2019-04-15:
I think you should join us in IRC by pressing the blue button in the GitHub
readme; this bug is strange. The symptoms suggest there's a communication
problem between the daemon and the keyboard.

MokeyII, 2019-06-27:
Was this issue resolved?
I am having the exact same issue with Corsair K95 
Also using a Corsair Scimitar RGB Pro 

Would love to find a fix for this so I can stay on linux.

tatokis, 2019-06-27:
Which Corsair K95 do you have? `lsusb -d1b1c:`

Make sure you are using the latest kernel and try a different USB host controller on your computer.
For example, if you are using USB2, try USB3 and vice versa. Also, if on a desktop try both the rear and the front ports.

Does ckb-next throw any errors? Can you generate and attach a report?
You might have some luck adding the usbcore quirks https://github.com/ckb-next/ckb-next/wiki/Troubleshooting#usbcore-quirks

hilumcool, 2019-07-23:
Hi

Yesterday I bought a Corsair K95 Platinum keyboard with a M65 Elite mouse and had the same trouble. to test the hardware, I connected the keyboard to a Windows machine and installed the latest iCUE software. The keyboard worked like a charm on Windows. Therefore I inspected the different options of iCUE. I came across an option like "poll rate" (sorry, my GUI is not in English, so the label in iCUE may be different). I changed the value to 8ms. (this value is now also shown in the ckb-next gui -> device tab)

Then I reconnected the keyboard to a USB3 connector of my linux machine (Raspberry Pi 4B) and... after a reboot everything worked as expected.

output of `lsusb -d1b1c:`
Bus 001 Device 004: ID 1b1c:1b5a Corsair 
Bus 001 Device 003: ID 1b1c:1b2d Corsair 

Maybe this helps. I'm not sure, if this value can be set with ckb-next and whether all K95 keyboards behave the same way.