ckb-next/ckb-next #399: opens in the worlds biggest window

c0ldn3s5, 2019-04-16:
opens in a window that is huge,lags everything. i had it installed on kali kde and it was great, even made it run as root. now im just on the regular kali linux. I didnt even run this one as root and its doing this. what do u need from me to help yall fix this because if i fix it on my own i wont tell yall

#something-else

c0ldn3s5, 2019-04-17:
[ckb-next-report-1555467608.gz](https://github.com/ckb-next/ckb-next/files/3087661/ckb-next-report-1555467608.gz)

i have narrowed it down to being caused by having two monitors, also might have the fix.

c0ldn3s5, 2019-04-17:
i fixed mine by unhooking the hdmi fromt he other monitor, opened ckb-next, minimized and maximized it, resized and maxed it again. Plugged in 2nd monitor and it stayed the same size. YW, i do what i can. let me know what i can do ty

c0ldn3s5, 2019-04-17:
also im not running root for the first time in my life on linux, im making an active change in my life all around it all... lol

c0ldn3s5, 2019-04-17:
like idk where to make request but if anyone does the dark core rgb let me know thanks

tdis8629, 2019-04-17:
Could you take a screenshot of the window? It sounds like it's an issue with HiDPI in GNOME.

c0ldn3s5, 2019-04-18:
> Could you take a screenshot of the window? It sounds like it's an issue with HiDPI in GNOME.

I can assure you its a multiple monitor issue but im only good at breaking things lol.
[image]

c0ldn3s5, 2019-05-17:
Reinstalled ckb on a fresh installation of debian this time with two monitors going and this problem did not persist. I will mark closed within a couple of days I suppose?..

closed 2019-05-20 by c0ldn3s5