ckb-next/ckb-next #401: CMake Error While Installing cannot find target directory

TLable, 2019-04-21:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.

This issue template has 3 main sections of which you should use only 1:
- [New Device Support Request](#new-device-support-request),
- [Feature Request](#feature-request), or
- [Bug Report](#bug-report).

If your issue doesn't fit in any of those categories, please use the [#something-else] section at the bottom.
Delete the redundant sections before sending.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

# Corsair STRAFE RGB Keyboard.

Enter the name of your device here and attach the report generated above.

---

# Feature request

## When I install I get no device detected, and after attemping other version I get a CMake error.?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## It should finish installing quickinstall file, which I am greatful whom put the time and thankful for all the hard work to create a compiled builder. The version that works installs but misses my keyboard  driver and after clearing the sys of the older files and attempting install of all dependencies and new version of cbk-next-master I return with 
[ 50%] Built target mviz
[ 51%] Automatic MOC and UIC for target ckb-next
[ 51%] Built target ckb-next_autogen
[100%] Built target ckb-next
Install the project...
-- Install configuration: "Release"
-- Up-to-date: /usr/local/bin/ckb-next-dev-detect
-- Installing: /usr/local/bin/ckb-next-daemon
CMake Error at src/daemon/cmake_install.cmake:65 (file):
  file INSTALL destination: /usr/lib/systemd/system is not a directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:43 (include)
  cmake_install.cmake:42 (include)

## Installing with the required drtivers but keep getting an error when building. I've installed it a few times and uninstalled it because the keyboard was not being detected. I have a STRAFE RGB  keyboard.
This is the error when installing. 

CMake Error at src/daemon/cmake_install.cmake:65 (file):
  file INSTALL destination: /usr/lib/systemd/system is not a directory.

## Looking into CMake to have checked its dir is not the issue. It may be possible that its using the wrong version of a dependency or has missing dependencies. Or could be the driver.  

---

# Something else

tatokis, 2019-04-21:
Which distro are you running?
What does this command return?
```
ls -la /usr/lib/systemd/system
```

TLable, 2019-04-21:
$ cat /etc/os-release
NAME="Pop!_OS"
VERSION="18.10"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Pop!_OS 18.10"
VERSION_ID="18.10"
HOME_URL="https://system76.com/pop"
SUPPORT_URL="http://support.system76.com"
BUG_REPORT_URL="https://github.com/pop-os/pop/issues"
PRIVACY_POLICY_URL="https://system76.com/privacy"
VERSION_CODENAME=cosmic
UBUNTU_CODENAME=cosmic

I return the line:
-rw-r--r-- 1 root root 201 Apr 19 23:54 /usr/lib/systemd/system

tatokis, 2019-04-21:
What about
```
sudo cat /usr/lib/systemd/system
```

TLable, 2019-04-21:
returns line:
[Unit]
Description=Corsair Keyboard Daemon

[Service]
ExecStart=/usr/bin/ckb-daemon
Restart=always
StandardOutput=syslog
StandardError=inherit
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target

tatokis, 2019-04-21:
If you delete it `sudo rm /usr/lib/systemd/system` and then re-run quickinstall. What happens?
Does the file come back? `sudo cat /usr/lib/systemd/system`

TLable, 2019-04-22:
Do you know keyboard shortcuts to search for /use/lib/systemd/system/

tatokis, 2019-04-22:
If you run `sudo cat /usr/lib/systemd/system` it will either show its contents if it exists, or it will say No such file or directory.

TLable, 2019-04-22:
I think I've got a dual architecture failsafe to stop important files from
being removed, cause I've added architecture 'i386' & '1386' and then I run
the line returns:
RM: cannot remove '/usr/lib/systemd/system/': Not a directory

TLable, 2019-04-22:
Y es it is both not a directory when remove is used 'Not a directory' &
running a daemon when searched for contents
[Unit]
Description=Corsair Keyboard Daemon

[Service]
ExecStart=/usr/bin/ckb-daemon
Restart=always
StandardOutput=syslog
StandardError=inherit
RemainAfterExit=yes

[Install]
WantedBy=multi-user.target

tatokis, 2019-04-22:
Please run the commands exactly as I type them.
Try:
```
sudo rm /usr/lib/systemd/system
sudo mkdir /usr/lib/systemd/system
```

Do note that there is no `/` in the end of either of those commands.

After that, quickinstall should work normally.

TLable, 2019-04-22:
Could you tell me a shortcut to locating those files? For example if I
wanted to find cosmic dir.

Thank you Tasos. I ran both code for ckb the quickinstall next and is
working fine.

TLable, 2019-04-22:
Have you used Origin in the past?

TLable, 2019-04-25:
I attemnpted to remove the file with the suspect suspect architecture arch.(?), which also included:
i386
1386
System now seems to have cleared the calls of errors while updating.

closed 2022-10-03 by tatokis