ckb-next/ckb-next #402: System Extension Not Allowed still appearing in cbd-next even though Karbiner Elements is already installed.

Ninja73737, 2019-04-21:
Do I have to uninstall Karbiner Elements to install this software properly? And if that is the case can I reinstall it afterwards? Also, I don't see a way in the karbiner elements software to back up my profile, I could be missing it but otherwise I will have to recreate all the rebindings I have set up. Let me know if there is an alternative.

ZirconiumX, 2019-04-21:
CKB uses the kernel extension from Karabiner Elements, but conflicts with
Karabiner because otherwise both programs would grab the device. You need
to uninstall Karabiner Elements to allow CKB to use your keyboard. You
might also need to enable the kernel extension in System Preferences.

>

Ninja73737, 2019-04-21:
I'm actually just using the software with a mouse. Is there any way the
software can both be used at once with separate devices? For instance if
I'm only controlling the internal keyboard with karabiner, and I'm using
ckb-next to control my mouse would that cause a conflict?

ZirconiumX, 2019-04-21:
Yeah, because there's only the one driver, sadly.

closed 2020-02-17 by Ninja73737