ckb-next/ckb-next #405: ckb overrides setxkbmap

jorne66, 2019-04-23:
ok so the issue is i set following command in .profile: setxkbmap -option caps:shiftlock
but gets overridden by ckb.
there is also no option to remap caps to shift lock in ckb

steps to reproduce: run setxkbmap
                                install ckb
                                
also works when command is entered before ckb is started

tatokis, 2019-08-16:
Quite a bit late, but we do not mess with X or layouts. It's possibly an X/libinput/evdev bug.

closed 2019-08-16 by tatokis

bqv, 2025-09-01:
Still a bug.

bqv, 2025-09-01:
Here’s what’s likely happening:

ckb-next-daemon enumerates input devices when it starts and creates its own “virtual keyboard” device (through uinput or X11).

That device registers with X as an input source, and X applies a layout to it.

By default, X may apply the first available XKB layout or fall back to us if unspecified.

Some DEs and input methods (GNOME, KDE, fcitx, ibus) will then see a “new keyboard” and try to synchronize layouts across devices. That looks like your current layout just got reset.

So it’s not that ckb-next explicitly calls setxkbmap — it’s that the act of creating a new X input device can trigger X/DE/input method to reshuffle layouts.