ckb-next/ckb-next #406: Feature request

csakis, 2019-04-24:
I absolutely love this software. Well done, guys! I would like to suggest a feature that exists in the windows version of iCue. It would be nice to have an animation based on the temperature of the CPU and GPU. I would love to map my Esc key to that.

ZirconiumX, 2019-05-13:
I thought I'd responded to this, but apparently not.

Custom animations like this are essentially the reason the pipe animation exists, and you should be able to write a script using the output of lm-sensors and @fleischie's [documentation on the pipe animation](https://github.com/ckb-next/ckb-next/wiki/Animations#pipe).

Gaming4LifeDE, 2019-07-09:
you could still have a few easy presets