ckb-next/ckb-next #408: Disable GUI preview on focus loss

tatokis, 2019-04-28:
It would be a good idea to disable the gui preview dynamically, whenever the main window loses focus. Would need to be careful so that this happens only if the user focuses another application (and not a popup of the GUI).

Finally, there should be a message in the preview area indicating that the preview was disabled only because the window lost focus, in case the user moved the window to a visible area.

fleischie, 2019-05-16:
After a quick 5 minute internet research, I came up with the following two links:

- https://doc.qt.io/qt-5/qguiapplication.html#applicationStateChanged
- https://doc.qt.io/qt-5/qt.html#ApplicationState-enum

The most appropriate value for the `ApplicationState` is probably `Qt::ApplicationHidden`, because it says "[...] should free up all graphical resources when entering this state."

I don't know about `Qt::ApplicationSuspended` and/or `Qt::ApplicationInactive`, we currently don't mine for bitcoins in the background, do we? :trollface:

fleischie, 2019-05-18:
I might be mistaken, but `Qt::ApplicationHidden`, and `Qt::ApplicationSuspended` are primarily used on mobile platforms, where the OS has a stricter requirement on control over the application state. This and the fact that multiple different systems exclusively emit `Qt::ApplicationInactive` and `Qt::ApplicationActive` on state changes convinces me, that only the latter should show the preview.

fleischie, 2019-05-21:
Just a guick fyi: I have started working on this issue.
Current state: Preview toggling based on application state works.
Still missing: Message about disabled animation preview.

Will hopefully submit a PR before the end of time.

fleischie, 2019-05-22:
PR #418 submitted.

closed 2020-01-06 by tatokis