ckb-next/ckb-next #410: Corsair M65 Pro RGB being detected as a sdl2 gamepad/joystick device by Steam

viggy96, 2019-04-29:
# Feature request

## What is the program's current behavior? 
Corsair gaming mice appear to Steam as an sdl2 joystick/gamepad.

## What would you like it to do instead?
Not appear as an sdl2 joystick gamepad.

# Something else
Steam sees the mouse as a joystick/gamepad device, causing many games in Steam to crash on startup. With a regular mouse plugged in instead, all games work just fine. 
```
Local Device Found
  type: 1b1c 1b2e
  path: sdl://1
  serial_number:  - 0
  Manufacturer: 
  Product:      ckb1: Corsair Gaming M65 Pro RGB Mouse vKB
  Release:      306
  Interface:    -1

Local Device Found
  type: 1b1c 1b2e
  path: sdl://2
  serial_number:  - 0
  Manufacturer: 
  Product:      ckb1: Corsair Gaming M65 Pro RGB Mouse vM
  Release:      306
  Interface:    -1
```

ZirconiumX, 2019-04-29:
I thought we'd fixed/mitigated this with mattanger/ckb-next#357. Apparently not.

The problem is likely not with us directly, but with how libinput detects joysticks (by guessing). Which version of libinput do you have?

viggy96, 2019-04-29:
I've got libinput version 1.10.4-1 installed, on Ubuntu 18.04.2 LTS.

viggy96, 2019-04-30:
Removing the device via udev rules works, but then the mouse is unusable. The udev rule suggested in mattanger/ckb-next#357 doesn't work. Setting the mode to "0000" in the udev rule also has no effect. Steam still picks up the mouse as a controller.

viggy96, 2019-05-03:
Right now, my solution is to disable the ckb-next-daemon.

viggy96, 2019-05-03:
@ZirconiumX Has a solution been found for this?

tatokis, 2019-05-03:
It's most likely something in your end that undoes what our udev rule does.
As a sidenote, the M65 itself has an HID descriptor that shows up as a joystick without the daemon running.

You should try looking at the order your udev rules are processed, and see if there is any rule that may mark daemon devices as joysticks.

I'm on Ubuntu 19.04 and there's no issue. I also tried in an 18.04 VM.

```
tatokis@tatokis-PC:~/sdl-jstest/build$ ./sdl2-jstest --list
No joysticks were found
```

If you can't figure out which rule is causing it, then switch to evdev instead of libinput. It does not have this issue.

viggy96, 2019-05-04:
@tatokis My Corsair mouse does not appear as a joystick under '/dev/input' though, as in your other comment. It only appears to steam that way, and to sdl2-jstest.

viggy96, 2019-05-04:
@tatokis @ZirconiumX I've fixed it by moving the ckb-next udev rule from its default install location, '/usr/lib/udev/rules.d', to '/etc/udev/rules.d'. Now, when running sdl2-jstest, there are no joysticks detected.

viggy96, 2019-05-08:
I suspect that the udev rule should be installed in '/etc/udev/rules.d' instead of the current location, '/usr/lib/udev/rules.d' to properly prevent the mouse from being detected as a joystick/gamepad.

ZirconiumX, 2019-05-08:
Udev should check either folder, so it shouldn't make a difference, and
given that it was reported as fixed for multiple people, I'm inclined to
believe there's something wrong with your system compared to everybody
else's.

viggy96, 2019-05-08:
I have confirmed that rules in the '/usr/lib/udev/rules.d' are not executed at all. It appears on the Ubuntu manpage for udev that this location is not checked for udev rules: http://manpages.ubuntu.com/manpages/bionic/en/man7/udev.7.html

It has been checked in the past, for example in Ubuntu version 14.04, '/usr/lib/udev/rules.d' was a valid location for udev rules. But in 18.04, it doesn't appear that this is a valid location for installing udev rules.

viggy96, 2019-05-08:
Moving the 99-ckb-next-daemon.rule to '/lib/udev/rules.d/' caused the rule to be executed properly, as the manpage would suggest.

tatokis, 2019-05-09:
I'm not sure what we can do about this, then.
Ubuntu 16.04 also doesn't check `'/lib/udev/rules.d/'` and that's still supported.

Perhaps there's a udev/systemd version check we can do to find out when this change was done? I'm out of ideas.

viggy96, 2019-05-09:
@tatokis Really, Ubuntu 16.04 doesn't check `'/lib/udev/rules.d'`? Even though it says xenial is supposed to check there? Hmm... Interesting. This does seem to be a pesky issue. I was thinking that we could change the udev rule install location to `'/lib/udev/rules.d'`, and those on 14.04 would just have to upgrade, since that release's support end date has passed. Does the ckb-next-daemon rule not execute when you move it to `'/lib/udev/rules.d'` in 16.04?

tatokis, 2019-05-09:
I just realised. The rule is supposed to be installed in `/lib/udev/rules.d/` exactly because rules aren't being read from the /usr/lib directory.
https://github.com/ckb-next/ckb-next/pull/311#issuecomment-453755239

tatokis, 2019-05-09:
How are you installing the software?

viggy96, 2019-05-09:
How am I installing ckb-next? I clone the repo, and run the quickinstall script.

tatokis, 2019-05-09:
https://github.com/ckb-next/ckb-next/blob/fd509f5bb25aabbcc982bb28cb6f445a875648a0/src/daemon/CMakeLists.txt#L474

Well then... I told it to install to the wrong directory.

viggy96, 2019-05-09:
@tatokis LOL, things happen

viggy96, 2019-05-14:
Has this change been completed?

tatokis, 2019-05-16:
Just did

commit fe369d68 (2019-05-16): Switch udev rule installation path to /lib

On distros where /lib is symlinked to /usr/lib we do not try to
remove the old rule.

Closes #410
closed 2019-05-16 by tatokis