ckb-next/ckb-next #413: questions about functions and which keyboard to get for ckb-next

prankousky, 2019-05-19:
Hi everybody,

I am considering buying a new keyboard (currently considering the `Corsair K95` to use with `ckb-next`. I thought of the K95 because of those G keys, because I thought of mapping those for things like my main actions (G1 = `,N` for `:NERDTreeTabsToggle`, for example). However, I have some questions:

* will `ckb-next` work alongside `i3gaps`? I have multiple keybindings set up in my i3 config, mainly using `mod1` and `mod4`. Can it run alongside i3 without interfering with its keybindings **and** still be functional (for example, let me map different keys, change backlight colors, etc.)?

* can I do something like this: use `xprop` or similar to find out if the current terminal window contains `vim` in its name, and if so, change backlight colors for h,j,k,l; then when focusing some other window, change those colors back to their regular colors (this is probably not merely a `ckb-next` thing, but perhaps somebody here has wanted the same thing and might already have figured out how to)

* is it possible to create modes? For example, if I hold down `G2`, nothing should happen; but when I hold down `G2`, *then* press `i`, action X should be executed, if I press `o` instead, action Y should be executed happen? Or can I **only** bind pressing those G keys to one action, but not combine them with pressing other keys?

This was the software related part. Hardware wise: should I just go with the `Corsai K95`? It seems like a solid keyboard for a decent price. Or which one would you recommend? I considered the `K70` before, but it does not have those G Keys and having those would be such a great improvement on my workflow that I'd like to buy a keyboard that has extra keys like this. I doesn't have to be the K95, perhaps somebody here will recommend another one that has similar features.

Changing light color per key is important to me **if** that feature I have been asking above (tl;dr: coloring specific keys per focused program) is possible. I don't need those colors for the looks (even though they DO look great), but rather for extra functionality.

Thanks in advance for your input :)

ZirconiumX, 2019-05-19:
> Hi everybody,
>
> I am considering buying a new keyboard (currently considering the `Corsair K95` to use with `ckb-next`. I thought of the K95 because of those G keys, because I thought of mapping [...]

I have no idea. Since ckb-next operates at a lower level to `i3gaps`, everything will first go through it before it reaches your window manager. If you set a keybind in ckb-next, it will trigger before i3 ever sees it. This might mean it interferes, or it might not.

> * can I do something like this: use `xprop` or similar to find out if the current terminal window contains `vim` in its name, and if so, change backlight colors for h,j,k,l; then when focusing some [...]

ckb-next does not support this natively, but it supports some things which can be used to achieve what you suggest.

There is the [pipe animation](https://github.com/ckb-next/ckb-next/wiki/Animations#pipe). In this mode, ckb-next reads commands from a file in `/tmp`, and you can then write a script which writes [`rgb` commands](https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual#led-commands) to that file. In this case your script can poll `xprop` and then write something like `rgb h,j,k,l:ffffffff` to the file in `/tmp`, or else write `rgb h,j,k,l:000000`.

Alternatively, you can use a script for vim:
```
#!/usr/bin/env bash
ckb-next --mode vim
/usr/bin/vim $*
ckb-next --mode default # or whatever your default mode is called
````
This means you can set vim-specific keybindings to be used whenever you run Vim, but they'll always run while Vim is active.

> * is it possible to create modes? For example, if I hold down `G2`, nothing should happen; but when I hold down `G2`, _then_ press `i`, action X should be executed, if I press `o` instead, action Y [...]

Unfortunately, the macro system does not support what you are suggesting; it triggers on a single keypress only (in this case G2), not on combinations. While this would be more powerful, the macro code is a bit messy at the moment and we don't understand it very well.

> Hardware wise: should I just go with the `Corsair K95`? [snip] Or which one would you recommend?

There are three Corsair K95 keyboards; the K95 Platinum RGB (with 6 macro keys and hardware animation support), the K95 RGB (with **18** macro keys) and the K95 (non-RGB). We support all three, but you can't currently save to hardware or use the hardware animation support on the K95 Platinum (see #95).

@tatokis has a K70 RGB, and I have a Strafe RGB. Honestly, I haven't found the idea of macro keys all that useful.

> Changing light color per key is important to me **if** that feature I have been asking above (tl;dr: coloring specific keys per focused program) is possible. I don't need those colors for the looks [...]

prankousky, 2019-05-19:
Wow, thank you so much for that detailed reply.

So unless I use the macro keys in my i3 config, they should not interfere... this is already good, because then I can do what I planned on doing.

```
#!/usr/bin/env bash
ckb-next --mode vim
/usr/bin/vim $*
ckb-next --mode default # or whatever your default mode is called
```

So just to make sure I got that right: I could do this for any program I'd like to and have, for example, the macro keys behave differently for each program? Same with those rgb commands... "if vim is running, highlight hjkl, if mpv is running, highlight 9,0,s,space", etc... if this is the case, I'm almost already sold on ~~the K95~~ `ckb-next` and some supported keyboard.

> There are three Corsair K95 keyboards

Seems like the only one of those available in my country (Germany) is the `Corsair K95 RGB Platinum` with "only" 6 macro keys. At least I can't find the other ones you've mentioned on their official German website, nor on amazon (so I assume they just don't sell them here if they don't even display them on their website).

> but you can't currently save to hardware or use the hardware animation support on the K95 Platinum

As long as the daemon runs on my computer I won't need to save to hardware, correct (assuming I only want to use it on my computer)? Since I'll "have to" buy the K95 RGB Platinum (if I want the K95), I guess hardware animation support is not **that** important. Just to be sure: that means something like fading all keys from green to blue to yellow etc. will **not work**, _**but I could still tell**_ it "i'm in vim, color h,j,k,l until I exit  it" according to your example from above?

The K70 is available here as well. I don't know if that is just due to bad product images, but it looks like it has WAY better colors than the K95; is this the case? Huge downside imho is no macro keys, plus I prefer the K95s larger enter/return key to the K70s smaller one. 
Same with the Strafe, but it neither has macro keys, nor media keys (which would "force" me to "waste" keybindings on media control; unless I can use the `fn` key for this with i3, which I'll try after pposting this reply).

I have never had a keyboard with macro keys, but they seem really helpful to me. I have way too many keybindings, and just being able to put certain actions on those buttons seems really cool. I use win and alt for i3 (so `win+f` is full screen, `alt+m` enters "music mode" allowing me to control `mpd`). Things that aren't used often enough to remember (some vim keybindings, some modes I rarely use) would be perfect to just put on those keys and have a little cheat sheet or stickers telling me what which one does in which content/program. Then again, it is definitely not something really necessary, but if I'd have to chose between the K70 and K95 at the moment, I'd go with the K95 just to see whether or not I actually need those macro keys (because of similar price and features, unless the K95 really has way worse colors than the K70...)

tatokis, 2019-05-19:
>So unless I use the macro keys in my i3 config, they should not interfere... this is already good, because then I can do what I planned on doing.

Correct, by default, ckb-next just passes key events through to the kernel.

> So just to make sure I got that right: I could do this for any program I'd like to and have, for example, the macro keys behave differently for each program?

You can either send rgb data from inside the program using pipe as mentioned above, or switch profiles/modes that have colour schemes and bindings you set yourself, with some script that's watching what programs are running/are in focus. Do keep in mind that there is a slight delay when switching profiles outside the application.

> Seems like the only one of those available in my country (Germany) is the Corsair K95 RGB Platinum with "only" 6 macro keys.

The other two have been out of production for some time now. So if you want more macro keys you'll need a second hand one.

> As long as the daemon runs on my computer I won't need to save to hardware, correct (assuming I only want to use it on my computer)?

That is correct, save to hardware is needed for when there's no software. You can control the lights and key bindings as long as the daemon and the GUI are running.

> I don't know if that is just due to bad product images, but it looks like it has WAY better colors than the K95; is this the case?

They should both be pretty bright. Don't think there is a difference, but I could be wrong.

Do note that there is no way to pass through the G keys to the OS as G keys. You either use ckb-next to launch applications when they are pressed, or you bind them to F13, F14, and so on, so that you can then handle them using your existing setup.

ZirconiumX, 2019-05-19:
Mostly I agree with what Tasos said, but:

> So just to make sure I got that right: I could do this for any program I'd like to and have, for example, the macro keys behave differently for each program? Same with those rgb commands... "if vim [...]

You would have to use the pipe animation for this; profiles don't stack.
> Seems like the only one of those available in my country (Germany) is the `Corsair K95 RGB Platinum` with "only" 6 macro keys. At least I can't find the other ones you've mentioned on their official [...]

The non-platinums are not officially produced anymore, but I wasn't sure where you where getting your keyboard from, so I decided to play it safe.

> > but you can't currently save to hardware or use the hardware animation support on the K95 Platinum

> As long as the daemon runs on my computer I won't need to save to hardware, correct (assuming I only want to use it on my computer)? Since I'll "have to" buy the K95 RGB Platinum (if I want the [...]

Fading all keys will work perfectly fine, because it's software controlled by ckb-next, but when the daemon isn't running, you'll be stuck with whatever the default hardware profile is, and we currently offer no way of changing that.

> I have never had a keyboard with macro keys, but they seem really helpful to me. I have way too many keybindings, and just being able to put certain actions on those buttons seems really cool. I use [...]

I have very few keybinds in Openbox (I have bindings for the audio keys, alt-space brings up a terminal, and alt-menu brings up the logout screen), which might be why they're not appealing to me.

prankousky, 2019-05-19:
Thank you all. I just ordered the K95 Platinum and will see if I can get everything working.

prankousky, 2019-05-19:
Thank you all. I just ordered the K95 Platinum and will see what I can do with it once it gets here :)

prankousky, 2019-05-21:
I received my keyboard today and already love it and your program! It is the `K95 Platinum RGB`. Can I safely update to the recent firmware? I get info about this any time I start the GUI, but wasn't sure whether or not is is fine or if updating might break something...

Everything seems to work perfectly as far as I can tell. Getting on creating color modes for general use and different programs, this is really an awesome tool. Thank you so much for creating it.

tatokis, 2019-05-21:
Yes, it's safe. Even if something goes wrong you can easily recover it.

closed 2019-05-21 by tatokis