ckb-next/ckb-next #414: milestones?

BloodyIron, 2019-05-20:
Any chance we can get milestones used here on github so we know how close-ish the next release is?

There's a bug that was fixed about 19 days ago, and I don't know when the next release will happen so I can get the update and fix.

ZirconiumX, 2019-05-21:
Even if we used them, they wouldn't be a particularly good indicator. We
generally aim to release roughly every three months, but at the moment we
don't have much to give to people. That's mostly because we've burned out.

BloodyIron, 2019-05-21:
Why not just release a minor version update with the main commit branch? Not all releases need to be big ticket items. This doesn't have to be david vs goliath.

ZirconiumX, 2019-05-21:
You can always build it from source.

BloodyIron, 2019-05-22:
But you already have a build and release pipeline? Why not just roll out a release since that's already in-place? :/

I literally picked a mouse because of ckb-next and I didn't think burn-out was going to be an issue only a month after purchase...

ZirconiumX, 2019-05-22:
> But you already have a build and release pipeline? Why not just roll out a
> release since that's already in-place? :/
>
Uh, no? That "pipeline" is just @tatokis building a DMG for Mac, from
source, in a VM that sometimes breaks, and us writing release notes.
There's no automated system for this in any way.

> I literally picked a mouse because of ckb-next and I didn't think burn-out
> was going to be an issue only a month after purchase...
>

It was an issue about six months before you bought that mouse, and I think
we've been fairly transparent about that.

We're two university students, with occasional others, who have to support
everything from scratch by staring at a byte stream to guess what is
happening. We don't get support or assistance from Corsair.

It's what happens when people make demands of time and effort without
offering theirs in return, like telling us to make releases because this
particular randomly chosen commit has a fix they want.

It's entirely open source, and offered to the public out of the good of our
hearts. If you want a release, build one from scratch, or pitch in to help
us add support for things so that we consider it worth a release.

>
>

closed 2019-05-22 by tatokis