ckb-next/ckb-next #416: [Feature Request] Notifications when profiles change

thespyboss, 2019-05-21:
Can there eventually be a feature where you can have a notification when a profile is switched? That would be a great addition to ckb-next!

ZirconiumX, 2019-05-21:
I'm struggling to find a use case for this; the only way to change profiles
is manually, so notifying somebody that yes, they did just change the
profile feels a bit redundant. If someone wants it in a notification they
can invoke Zenity.

tatokis, 2019-05-21:
IMO It'd be useful if you have multiple similar profiles/modes and are switching between them using a keyboard button/key, since you can't see the name.

thespyboss, 2019-07-03:
I wonder how to set that up? I might start looking on how to set that up in a few days.

thespyboss, 2019-07-03:
And Also, @ZirconiumX it would be like that ICUE notification system. Like for me, I have a few profiles that are the same but have different animations. It would be nice to get like a small banner like the one in ICUE to know which profile i just switched to.

tatokis, 2019-07-03:
> I wonder how to set that up?

You will need to read about Qt slots and signals. Once you figure out how these work, if I'm not mistaken, one is emitted whenever a mode changes in the GUI. Keep in mind that each device has its own mode with its own name and information, as devices are handled separately.

You can connect a handler to the slot, and then over Qt's DBus bindings send a desktop notification.
https://doc.qt.io/qt-5/qdbusinterface.html
https://developer.gnome.org/notification-spec/

Make sure the notification part is modular (its own class), as it will eventually need to be implemented for mac and windows as well.

thespyboss, 2019-07-04:
What about attaching the notification system to the Change mode settings? The signal could be changed with the user's preference of which key would change the profiles.