ckb-next/ckb-next #419: connection lost when closing GUI

prankousky, 2019-05-23:
Sorry if this is a stupid question. I read through troubleshooting etc., but couldn't figure out how to fix this.

What I did so far

* install `ckb-next` from AUR
* (just yesterday) replaced it with `ckb-next-git` from AUR
* connected my keyboard
* started `ckb-next` => works

When I quit the `ckb-next` GUI, I got a disclaimer that the app will continue to run in the background. Perfect!

After reboot I noticed that I had the regular demo pattern on my keyboard. I checked and `ckb-next` was **not** running. When I ran it manually, it worked again (switched to my `normal` mode, successfully detected the keyboard, etc.). But when I closed the GUI, it switched back to the demo rainbow pattern from before.

I enabled and started `ckb-next-daemon`, which seems to always be active. But now my keyboard only works as expected when I run the GUI in the background; as soon as I close it, it will switch back to rainbow demo.

How can I fix this? It worked fine when I first installed the program, but now requires the GUI to run constantly. I am currently just launching it via `i3-msg` and put it on a usually empty workspace, but that doesn't seem like the best solution.

Thanks for your help.

tatokis, 2019-05-23:
The GUI is supposed to run in the tray. i3 doesn't automatically launch applications and doesn't obey the XDG autostart standard.

You can run the gui with `ckb-next --background` on startup so it doesn't draw a window by default

closed 2019-05-23 by tatokis

ZirconiumX, 2019-05-23:
The GUI tells the daemon what to do and how to animate; it needs to be
running for animations to happen because the daemon is dumb. It doesn't
know what an "animation" is, it just gets a series of RGB commands applied
to various keys on a keyboard. Hell, it barely even knows what a key is
(it's just a number with a name).

On the other hand, the GUI doesn't know what a USB device is. It doesn't
know how the Corsair protocol works, or what letter pressing does. It just
knows that it needs to listen for strings and tell the daemon to do things
like animations.

ckb-next is both the daemon and the GUI; you need both running for it to
work well.