ckb-next/ckb-next #421: Macro delay is not working

karlosss, 2019-05-28:
I am trying to set up a macro which shall do the following:
- press and release `h`
- wait for 500 ms
- press and release `a`
- wait for 500 ms
- press and release `baskspace`

My macro code looks like this (built with the help of the wiki [here](https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual)):
`+h,-h=500000,+a,-a=500000,+bspace,-bspace`

When I press the button to which the macro is bound, it is executed fine except for the delays which are just not taken into account at all. Am I doing something incorrect or it might be a bug? If the latter, I am ready to provide more information.

Thanks!

tatokis, 2019-06-06:
Apologies for the delay.
The issue is that delays are absolute, starting from zero. They are more like timestamps.

If I got it right, what you want to do should be:
`+h=0,-h=500000,+a=500100,-a=1000100,+bspace=1000200,-bspace=1000300`

karlosss, 2019-06-06:
Seems to be it, thanks!

By the way, does it mean that this park of the wiki is wrong and is worth correcting?

`macro g5:+d,-d,+e=5000,-e,+l,-l=10000,+a,-a,+y,-y=1000000,+enter,-enter` define a macro for `g5` with a 5,000us delay between the e down and e up actions. A 1,000us delay between l up and a down, a delay of one second (1,000,000us) after y up and before enter, and the global delay for all other actions.

frickler24, 2019-07-09:
@karlosss Do you use the macro gui? If yes then delays are relative, not absolute and please check "as typed" in the gui

tatokis, 2019-08-16:
The wiki refers to the daemon, not the GUI. They have different syntax rules as far as I know.

closed 2019-08-16 by tatokis