ckb-next/ckb-next #426: Set brightness through CKB daemon?

BnMcG, 2019-06-07:
# Something else

Is it possible to set the brightness of the keyboard/mouse through the CKB daemon? I had a look through the wiki page here (https://github.com/ckb-next/ckb-next/wiki/CKB-Daemon-Manual) and I can see you can set the RGB of each key/the whole device, but I'm unsure if brightness is set differently and couldn't see anything obvious on the page. 

My use case is that I'd like to be able to turn off the LEDs on the device and then turn them back on via a command, hopefully this is possible through the daemon? 

Devices are a K70 RGB Lux and a Sabre RGB, and I can change the brightness of the LEDs via the GUI app itself.

tatokis, 2019-06-07:
It's not possible through the daemon. It does not control the brightness.

If you want to turn off the LEDs on your devices, you can create a profile for both your devices with all the LEDs turned off, or the brightness set to 0%, and then switch to the profile by calling ckb-next --profile "Name of your blank profile"

closed 2019-06-07 by tatokis

BnMcG, 2019-06-07:
Is there a way I'm missing of turning the LEDs off? I've created an "On" profile with the LEDs set as required with animations etc, and an "Off" profile with the brightness set to 0%, but I'm finding when I run ckb-next to switch the profile, the profile switches but the brightness doesn't update (ie: if it's 100% the LEDs stay on but the colours/animations disappear in the "off" profile, or if it's 0% the profile switches but none of the LEDs turn on)

tdis8629, 2019-06-07:
I'm away form my keyboard at the moment, but I believe that the LED brightness is global unless you change it in the settings under "Set Brightness Per Mode."

tatokis, 2019-06-07:
Yes, it is global by default. `#000000` is basically the same as 0% when it comes to how ckb handles brightness.

Just keep in mind if you turn on per-mode brightness, then you won't be able to control your mouse's brightness using your keyboard's brightness button.

For now, I'd say a profile with `#000000` is the best option, however it might be a good idea to add a `--brightness` argument