ckb-next/ckb-next #427: Make under linux gentoo failed after set to new profile  17.1

lovelytux, 2019-06-08:
Hey,
this week gentoo switched to new profile 17.1. The 32-bit-packages are now orphaned. 

The ./quickinstall failed with:

[ 99%] Building CXX object src/gui/CMakeFiles/ckb-next.dir/ckb-next_autogen/EJRQKI7XPS/qrc_image.cpp.o
[100%] Linking CXX executable ../../bin/ckb-next
[100%] Built target ckb-next
./quickinstall: line 65: sudo: command not found
An error occurred,
press enter to exit.

fleischie, 2019-06-08:
Erm... hi @lovelytux. The error message indicates, that you are missing the `sudo` command. Which is an external program, that elevates the user's permission to make administrative changes.

Try to setup this command and try again, please.

lovelytux, 2019-06-09:
Oh, so sorry. Yes it was my mistake. Thanks for reply!

closed 2022-10-03 by tatokis