ckb-next/ckb-next #429: M65 mouse no beeing recognized

luanalbineli, 2019-06-11:
Hi guys,

I've installed ckb-next on my Linux Mint 19.1, but unfortunately it not recognize my M65 (non rgb) mouse. Searching on the known issues, I've found the `usb.h` constant file, and I didn't found my idProduct: `0x1b05`.

I've also tried another usb ports (3.0), also unsuccessfully.

## General information

### Source
  * built manually using quickinstall script

## Logs & crash reports

Attach:
ckb-next-dev-detect - Tue Jun 11 15:43:30 -03 2019

Kernel: Linux luan-Predator-PH315-51 4.15.0-51-generic #55-Ubuntu SMP Wed May 15 14:27:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	LinuxMint
Description:	Linux Mint 19.1 Tessa
Release:	19.1
Codename:	tessa

USB Devices:
Couldn't open device, some information will be missing

Bus 001 Device 006: ID 1b1c:1b05 Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b05 
  bcdDevice            2.26
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
[... 130 lines cut ...]
Jun 11 15:41:39.492177 luan-Predator-PH315-51 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
989
/dev/input/ckb0/version
0.4.0-27-g10e6b54
/dev/input/ckb0/connected

Environment:
QT_QPA_PLATFORMTHEME=qt5ct
QT_ACCESSIBILITY=1
XDG_SESSION_DESKTOP=cinnamon
XDG_CURRENT_DESKTOP=X-Cinnamon
QT_BEARER_POLL_TIMEOUT=-1

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-4.15.0-51-generic root=UUID=640976a4-0f93-4f6c-848c-dee70720b392 ro quiet splash vt.handoff=1

## What is the program's current behavior?
It is showing a "No connected devices" message.

## What is the expected behavior of the program?
Show the M65 tab configuration?

tatokis, 2019-06-13:
The M65 non RGB is a completely different device from the RGB one, with its own protocol.
It is not high on the priority list, and probably won't be supported for a while.

If you want to try something, replace the usb id in usb.h of the `M95` with yours and see if that does anything.

luanalbineli, 2019-06-13:
Hi, tatokis, thanks for the reply.
This device is supposed to be supported, don't? At least it is listed in the Supported Hardware list.
About the change, is it just replace the usb id and run the `./quickinstall` script again? Or I need to compile something else?

ZirconiumX, 2019-06-13:
We support the M65 RGB, M65 Pro RGB and M65 Elite RGB. The RGB there means
it uses a specific protocol that we support. The M65 non-RGB does not use
this protocol, so we cannot support it.

luanalbineli, 2019-06-13:
Hi guys, thanks for your replies.

I've made the changes, as @tatokis sugested, and it partially worked. The DPI previously configured in the Windows driver was applied, but the configuration is "read only". The changes that I made in ckb does not reflect in the mice.

@ZirconiumX , about the support, the non RGB mice is listed as supported:
- M65:
  - non-RGB <-------------
  - PRO RGB

Anyway, it already helped me a lot, thanks.

tatokis, 2019-06-14:
Yeah, that's a mistake. Apologies, I'll fix it ASAP.

Regarding the mouse, if it does in fact share the same protocol as the M95, you can try adding this quirk in your linux kernel cmdline (through your bootloader), and see if you can actually write the configuration to the mouse, like changing DPI.

`usbcore.quirks=1b1c:1b05:gn`

Strazzom, 2020-03-10:
> If you want to try something, replace the usb id in usb.h of the `M95` with yours and see if that does anything.

I would like to report that I got the M65 working using this method. It does seem to use the same protocol, I can change the DPI successfully (it is somewhat janky though, it needs proper support). At the moment, I have it being recognized as the M95.

Saving the config to device causes it to stop working when I boot into Windows with it, and the mouse does not work unless the ckb-next program is started first. The DPI buttons are also misaligned (the highest one is interpreted as the sniper button).

gravityfargo, 2023-09-18:
Stale

closed 2023-09-18 by gravityfargo