ckb-next/ckb-next #430: CKB-NEXT missing images, not running animations

bab5470, 2019-06-13:
# Bug report

## General information

### Source
  * built manually using quickinstall script

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behavior?
I have a K55 RGB Keyboard and Harpoon RGB mouse. 

If I go to the harpoon tab: 
- And turn off the light for the mouse, it turns it off for the keyboard too.
- And  I select the rainbow mode I don't actually see the mouse light change color.

If I go to the K55 RGB tab:
- And  select *any* of the modes on the K55 RGB tab nothing happens (the colors on the keyboard remain a solid white/blue color)  despite selecting: Rainbow, or Breathing or Trippy. 

Additionally the lighting tab appears to be missing an image. I am assuming I should see a keyboard? Instead I see a bunch of squares. 

I have uninstalled and re-installed the application several times as well as the prerequisites. 

## What is the expected behavior of the program?
Selecting one of the other modes should change the lighting on the keyboard
Selecting rainbow mode on the mouse should use a rainbow.
There should be an image on the lighting tab of the K55 keyboard
Turning off the mouse light shouldn't also turn off the keyboard light.

## Any other notes & comments?
N/A
[ckb-next-report-1560421446.gz](https://github.com/ckb-next/ckb-next/files/3285483/ckb-next-report-1560421446.gz)

# Something else
N/A

tatokis, 2019-06-14:
> And turn off the light for the mouse, it turns it off for the keyboard too.

That is by design, it's so that you can control the brightness of the mouse with the brightness button on your keyboard. If you want to disable it, enable per-mode brightness.

> Additionally the lighting tab appears to be missing an image. I am assuming I should see a keyboard? Instead I see a bunch of squares.

There is no image. Typically, the keyboards have per-key lighting, so each key is its own square.
In your case, the K55 only has three lighting zones, and this was the easiest way to implement it.
You are welcome to try to make it look nicer.

> And select any of the modes on the K55 RGB tab nothing happens (the colors on the keyboard remain a solid white/blue color) despite selecting: Rainbow, or Breathing or Trippy.

That should've been fixed. Perhaps the animations aren't loading.

If you quit the GUI (Settings->Quit) and then run it from a terminal by typing `ckb-next`, can you paste the output?

bab5470, 2019-06-15:
Running ckb-next via terminal results in the following output:

`$ ckb-next
QObject::startTimer: Timers cannot have negative intervals
qt5ct: using qt5ct plugin
Gtk-Message: 07:25:57.459: Failed to load module "canberra-gtk-module"
Using QSytemTrayIcon
Path to settings: "/home/myuser/.config/ckb-next/ckb-next.conf"
qt5ct: D-Bus system tray: yes
Scanning "/usr/local/libexec/ckb-next-animations/gradient"
Scanning "/usr/local/libexec/ckb-next-animations/heat"
Scanning "/usr/local/libexec/ckb-next-animations/invaders"
Scanning "/usr/local/libexec/ckb-next-animations/life"
Scanning "/usr/local/libexec/ckb-next-animations/mviz"
Scanning "/usr/local/libexec/ckb-next-animations/pinwheel"
Scanning "/usr/local/libexec/ckb-next-animations/pipe"
Scanning "/usr/local/libexec/ckb-next-animations/rain"
Scanning "/usr/local/libexec/ckb-next-animations/random"
Scanning "/usr/local/libexec/ckb-next-animations/ripple"
Scanning "/usr/local/libexec/ckb-next-animations/snake"
Scanning "/usr/local/libexec/ckb-next-animations/wave"
Starting "/usr/local/libexec/ckb-next-animations/wave"
Starting "/usr/local/libexec/ckb-next-animations/gradient"
Starting "/usr/local/libexec/ckb-next-animations/gradient"
Downloaded new firmware list. 38 entries found.
Starting "/usr/local/libexec/ckb-next-animations/gradient"
Starting "/usr/local/libexec/ckb-next-animations/random"
Starting "/usr/local/libexec/ckb-next-animations/ripple"
`
It's not only are the animations not loading but again as I click on any of the pre-sets nothing changes on the keyboard itself - as long a s ckb-next is running the keyboard is a solid blue color no mater what animation I select. 

As soon as I exit though the rainbow animation resumes (I understand this is be cause ckb-next can't save to the hardware on the K55 RGB model)

bab5470, 2019-06-15:
> That is by design, it's so that you can control the brightness of the mouse with the brightness button on your keyboard. If you want to disable it, enable per-mode brightness.

Awesome - that did what I wanted!

bab5470, 2019-06-15:
There is a secondary issue - when running ckb-next the mouse cursor is insanely fast, when I close it is slows down substantially. With it stopped its a bit too slow, with it started its way too fast. 

Is there anything I can adjust in ckb-next to adjust the mouse speed or should I rely on something like gnome settings to adjust it?

ZirconiumX, 2019-06-15:
That's DPI; look at the DPI tab to change your mouse speed.

bab5470, 2019-06-15:
Regarding the error:
> Failed to load module "canberra-gtk-module"

I solved that with:
`sudo apt install libcanberra-gtk-module libcanberra-gtk3-module`

(It **doesn't** fix any of the issues mentioned above though)

bab5470, 2019-06-15:
> That's DPI; look at the DPI tab to change your mouse speed.
> […](#)

Thanks - I selected one of the other harpoon rgb mouse presets, and my mouse speed returned to its normal slower level. So I must have adjusted one of the DPI settings inadvertently.

bab5470, 2019-06-16:
> That should've been fixed. Perhaps the animations aren't loading.

I'm guessing (or at least my experience seems to be that it's not. Not only are animations not loading in the GUI but the color scheme doesn't take effect on the keyboard either. I just get a solid blue/white color on the keyboard when I open ckb-next and it doesn't change no mater which preset I select.

tatokis, 2019-06-24:
Apologies, I thought I responded to this.
Try selecting each animation for a given mode, then select all the rgb zones and press "set keys".
See if that fixes them

bab5470, 2019-06-25:
That works! Thanks!

closed 2019-06-25 by bab5470