ckb-next/ckb-next #431: Not detecting USB/Keyboard in Mac OS Catalina beta

MikesTooLz, 2019-06-14:
I installed the new Mac OS Catalina beta and CKB-Next no longer can detect my corsair mouse and keyboard. I know that this version of Mac OS is beta and just wanted to post this in here as a head up for you guys. Things I have already tried:

-Re-Installed
-Rebooted several times
-under (System Preverences > Security & Privacy > Input Monitoring) I tried adding ckb-next  in the allowed apps, then rebooted.
-unplugged and replugged USB devices and moved them around to different ports.

tatokis, 2019-06-14:
Of course they broke things again.
Generate and attach a report, please

MikesTooLz, 2019-06-14:
Report has been added, sorry about that.

tatokis, 2019-06-14:
Can you send a screenshot of the input monitoring menu?

MikesTooLz, 2019-06-14:
[image]

tatokis, 2019-06-14:
I meant `Security & Privacy > Input Monitoring`

MikesTooLz, 2019-06-14:
[image]
[ckb-next-report-1560475132.gz](https://github.com/ckb-next/ckb-next/files/3288628/ckb-next-report-1560475132.gz)

OK, I added it back into the input monitoring and then quit/relaunched ckb-next here is the screen shot and another report.

tatokis, 2019-06-14:
First, try unplugging and replugging one of your corsair devices.
I don't know how this works, but using the `+` can you add a specific program based on a path?
If so, try adding `/Library/Application Support/ckb-next-daemon` to the list of applications and then unplug/replug your devices again.

See if that helps

MikesTooLz, 2019-06-14:
BINGO!!!!!

I added `/Library/Application Support/ckb-next-daemon` to the list, then unplugged / replugged keyboard & mice but nothing happened. I rebooted the system and now it'
[image]
s working again.

tatokis, 2019-06-14:
That's excellent.

Can you generate another report? I want to see if the error changed after you unplugged and replugged

MikesTooLz, 2019-06-14:
[ckb-next-report-1560476235.gz](https://github.com/ckb-next/ckb-next/files/3288659/ckb-next-report-1560476235.gz)

Thanks a lot.
You got a PayPal address I can send a few bucks your way?

tatokis, 2019-06-14:
You're welcome. I'll have to come up with something so that people in the future can know what to do in order to get it working, preferably before 10.15 comes out. The error code doesn't seem to have changed.

I'll leave this issue open

If you want, you can use `tatokis [ a t ] windowslive [ d o t ] com`. Of course replace the the at and the dot.

MikesTooLz, 2019-07-04:
FYI, after todays beta update, I had to do this all over again.

atylmo, 2019-07-20:
I'm also having issues with the beta. I'm using a KVM switch (which works fine with two Windows PCs) and in the previous betas I couldn't use the keyboard at all without rebooting while the KVM was switched to the Mac. Sometimes ckb-next was able to see it. With the current beta I can decently reliably use the keyboard, but ckb-next won't detect it. It shows up in System Information under USB, though.

I got it to work once by adding the daemon and then also running the commands from https://github.com/ckb-next/ckb-next/issues/49#issuecomment-428004748 but it stopped again.

My keyboard is an original STRAFE RGB.

ItsEcholot, 2019-09-28:
[image]

I can't seem to find a way to add the `/Library/Application Support/ckb-next-daemon` to the Input Monitoring section in the latest beta version.

tatokis, 2019-09-28:
Not even if you click on the lock? Maybe you can generate and attach a report?

ItsEcholot, 2019-09-28:
Managed to get the + to appear after installing a software that correctly requests that permission (BetterTouchTool) and after adding the `ckb-next-daemon` there everything works again.

tatokis, 2019-09-28:
I didn't realise we needed to somehow request permission in a specific way.
I thought it happened the moment an application tried to open an HID device.

I'll have a quick look to see if I can figure out how/why this happens.

ItsEcholot, 2019-09-28:
Maybe because this is a root level daemon it behaves differently just a thought.

tatokis, 2019-09-28:
I think this is exactly why it happens. I'll try to make a workaround, but I can't test it

ItsEcholot, 2019-09-28:
Would be up for testing it if it helps

tatokis, 2019-10-04:
Quick unrelated question. To those of you who have tested ckb-next in Catalina, how did you get it to run? Did you have to run any commands, or did it just work like normal?

From what I understand we need to pay apple to get out software "notarized" before it can run on Catalina (which might be the end of ckb-next for mac), and I am trying to figure out what is going on.

https://developer.apple.com/news/?id=09032019a

ItsEcholot, 2019-10-05:
Besides this issue it worked like normal but maybe this is also because I already had it installed before I upgraded. The only thing I noticed that ckb sometimes wouldn't start the animations again after the keyboard got removed and then plugged in again (would display though that it's connected).

As far as I know the notarisation part only applies when you sign your application. And even then I think you can grant an exception in the security & privacy section of the Settings app when you run an unnotarised application.

tatokis, 2019-10-06:
@ItsEcholot thank you for your response. I guess we'll have to find out.

In the meantime, I found out that the way to get the daemon to be authorised is to create a launch agent and start the agent as the current user. Then, the prompt pops up, and once you allow access, you have to restart the daemon running as root.

Since the daemon is designed to only run as root, I made a workaround that lets it start as a normal user, ask for hid permission, and then quit once it receives it.

I'll implement this in the GUI and it should be good to go.

[image]

Vermyndax, 2019-10-07:
I just installed Catalina and a fresh build of ckb-next, and surprise... I was bit by this one. Did a new release with this fix get pushed through yet?

tatokis, 2019-10-07:
Still working on it. Will most likely have a new release tomorrow/soon-ish.

tatokis, 2019-10-08:
@Vermyndax @ItsEcholot Please try this build http://ckb-next.info/mac/ckb-next_v0.4.1-30-gd4ed59e_10.12-SDK.dmg

```
4b0947dcb4e0b58624ad3cb63ae08d9c18f8af8eba83cf0203284715a7f9e336  ckb-next_v0.4.1-30-gd4ed59e_10.12-SDK.dmg
```

If it works okay, I'll merge the code to master and make a new release tomorrow.

Vermyndax, 2019-10-08:
That worked for me, thanks!

jeffqchen, 2019-10-08:
> @Vermyndax @ItsEcholot Please try this build http://ckb-next.info/mac/ckb-next_v0.4.1-30-gd4ed59e_10.12-SDK.dmg
>
> ```
> 4b0947dcb4e0b58624ad3cb63ae08d9c18f8af8eba83cf0203284715a7f9e336 [...]

This one worked for me. Thanks for the quick respond.

ItsEcholot, 2019-10-08:
> @Vermyndax @ItsEcholot Please try this build http://ckb-next.info/mac/ckb-next_v0.4.1-30-gd4ed59e_10.12-SDK.dmg
>
> ```
> 4b0947dcb4e0b58624ad3cb63ae08d9c18f8af8eba83cf0203284715a7f9e336 [...]

Can confirm this works. ckb correctly requests the permission. The daemon restart command that gets executed seems to have a weird timing (>5 secs. after I plugged in the keyboard). Because of this delay I at first thought something went wrong or that I had to reboot manually. In the end it worked though.

tatokis, 2019-10-08:
Unfortunately the GUI can only check if the daemon has permission every >10 seconds due to launchctl. That's why you are getting that delay. I could not work around it.

ItsEcholot, 2019-10-08:
Ah I see makes sense. Well in that case it's for sure good enough for now

tatokis, 2019-10-08:
Any idea how to clear the database of the allowed applications?

Just as I am about to release 0.4.2, I can't get the prompt to show up anymore, and ckb-next-daemon says there's already access when the application isn't even listed under Input Monitoring.

tatokis, 2019-10-08:
Never mind that I guess. It started working now. No idea what changed.

ItsEcholot, 2019-10-08:
That's actually all it took for me, removing the ckb-next-daemon from the Input Monitoring section. Not sure if I had to restart the daemon or replug the keyboard but it appeared after that.

tatokis, 2019-10-08:
Just to verify. Please remove ckb-next-daemon from the list, then click the uninstall ckb-next button under settings.

Once you've done that, try this build and tell tell me if it works.
http://ckb-next.info/mac/ckb-next_v0.4.1-34-gea54480_10.12-SDK.dmg

```
c9d1b5e8a5977b8a87c6800be69c246b00b132f41bb9ef4514644ac8af7525c0  ckb-next_v0.4.1-34-gea54480_10.12-SDK.dmg
```

ItsEcholot, 2019-10-08:
Worked perfectly! Good idea adding that information about having to wait for the daemon restart.

Lemz97, 2019-10-08:
Im having the same issues, when i go into security and privacy the plus sign doesnt show in input monitoring

tatokis, 2019-10-08:
Click the uninstall ckb-next button. Reboot, then download and install https://github.com/ckb-next/ckb-next/releases/download/v0.4.2/ckb-next_v0.4.2.dmg

Lemz97, 2019-10-09:
ckb-next-daemon doesn't show in the security and privacy

tatokis, 2019-10-09:
Can you generate and attach a report please?

tatokis, 2019-10-09:
@ItsEcholot I thought you might be interested
[image]

ItsEcholot, 2019-10-09:
Interesting it stayed in the list for me but now my device doesn't appear in ckb-next anymore not sure what changed over night...

[ckb-next-report-1570626964.gz](https://github.com/ckb-next/ckb-next/files/3707507/ckb-next-report-1570626964.gz)

Also not sure if this has anything to do with Catalina but it seems the stability was greatly reduced since I made the os update. It seems it doesn't recognise devices anymore after standby but I will investigate this further and generate a report when I encounter it again.

kalanimatthews, 2020-02-25:
I have this exact problem. I did Catalina upgrade with no issue. I have 0.4.2  ckb. Today it no longer sees my keyboard and can't add to input monitoring like the above poster.
[image]

Ricardonacif, 2020-03-22:
Hey guys,

I have a fresh install of MacOS Catalina (10.15.3 (19D76)). The "Input Monitoring" tab on Security & Privacy also doesn't show anything. I've tried reinstalling it and some other things mentioned here, but no success. I've added a report.

Thanks,

[ckb-next-report-1584912166.gz](https://github.com/ckb-next/ckb-next/files/4366379/ckb-next-report-1584912166.gz)

kalanimatthews, 2020-03-23:
@Ricardonacif , I just migrated to my new mac with 10.15.3. Had the same problems. This is what I did that fixed it: 
1. Open System Preferences > Security and Privacy > General. I unplug/replugged keyboard with Sys Pref window open and it asked for me to allow the software first in General window, so I did.
2. In System Preferences, switch to Security & Privacy > Privacy tab > Input Monitoring. Mine showed nothing.
3. Quit ckb-next and restart it. Now it popped up in the Input Monitoring window. I allowed it then had to log into the console screen launched by the script. 
4. Enjoy - this worked for me and is working like normal now.

xguse, 2020-03-24:
For what its worth wrt @kalanimatthews note above:

Just unplugging and replugging as in the written step 1 was not enough for me. I had to go to System Prefs > Keyboard and click `Change Keyboard Type...` to get the rest of the process to work.

But it does seem to work! Thank GOD.

I am working at home and the damn wavy breathing effect was seriously detracting from my productivity. But then I am pretty ADD.

Anywho. Stay healthy y'all.

closed 2022-06-27 by tatokis