ckb-next/ckb-next #433: K70 RGB is not detected

thiagousa, 2019-06-18:
# Before you proceed

- Read the [wiki](https://github.com/ckb-next/ckb-next/wiki).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.
- Use [Markdown](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet) to format the message.

This issue template has 3 main sections of which you should use only 1:
- [New Device Support Request](#new-device-support-request),
- [Feature Request](#feature-request), or
- [Bug Report](#bug-report).

If your issue doesn't fit in any of those categories, please use the [#something-else] section at the bottom.
Delete the redundant sections before sending.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---

# New device support request

Enter the name of your device here and attach the report generated above.

---

# Feature request

## What is the program's current behavior?

## What would you like it to do instead?

## Any other notes & comments?

---

# Bug report

## General information

### Source

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behavior?

## What is the expected behavior of the program?

## Any other notes & comments?

---

# Something else

thiagousa, 2019-06-18:
lsusb | grep 1b1c | awk '{ split($6, a, ":"); print a[1] ":" a[2] ":gn"}' | tr '\n' ',' | sed -e 's/,$/\n/;s/^/usbcore.quirks=/'

usbcore.quirks=1b1c:0a14:gn,1b1c:1b36:gn,1b1c:1b3c:gn

no devices connected

tatokis, 2019-06-18:
> Click the Generate Report button in the Settings tab of the ckb-next user interface and attach the generated report to this issue.

thiagousa, 2019-06-18:
[image]
[image]

thiagousa, 2019-06-18:
I don't find this option.

thiagousa, 2019-06-18:
systemctl status ckb-daemon
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/lib/systemd/system/ckb-daemon.service; enabled; vendor prese
   Active: active (running) since Mon 2019-06-17 22:43:35 EDT; 19min ago
 Main PID: 5534 (ckb-daemon)
    Tasks: 1 (limit: 4915)
   Memory: 728.0K
   CGroup: /system.slice/ckb-daemon.service
           └─5534 /usr/bin/ckb-daemon

Jun 17 22:43:35 thiago-desktop systemd[1]: Started Corsair Keyboard Daemon.
Jun 17 22:43:35 thiago-desktop ckb-daemon[5534]:     ckb: Corsair RGB driver bet
Jun 17 22:43:35 thiago-desktop ckb-daemon[5534]: [I] Root controller ready at /d
lines 1-12/12 (END)...skipping...
● ckb-daemon.service - Corsair Keyboard Daemon
   Loaded: loaded (/lib/systemd/system/ckb-daemon.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-06-17 22:43:35 EDT; 19min ago
 Main PID: 5534 (ckb-daemon)
    Tasks: 1 (limit: 4915)
   Memory: 728.0K
   CGroup: /system.slice/ckb-daemon.service
           └─5534 /usr/bin/ckb-daemon

Jun 17 22:43:35 thiago-desktop systemd[1]: Started Corsair Keyboard Daemon.
Jun 17 22:43:35 thiago-desktop ckb-daemon[5534]:     ckb: Corsair RGB driver beta-v0.2.6
Jun 17 22:43:35 thiago-desktop ckb-daemon[5534]: [I] Root controller ready at /dev/input/ckb0

tatokis, 2019-06-18:
Maybe you should install ckb-next instead of ckb. No wonder your keyboard doesn't work with ckb

thiagousa, 2019-06-18:
@tatokis  Thanks it's works.

tatokis, 2019-06-18:
good to hear!

closed 2019-06-18 by tatokis