ckb-next/ckb-next #435: M65 Pro RGB mouse freezes on click after reboot

taisph, 2019-06-19:
# Bug report

## General information

### Source

Built from 10e6b54615dd8b300103b18738fea6d1c433e7d0 and installed using quickinstall.

## Logs & crash reports

[ckb-next-report-1560925757.gz](https://github.com/ckb-next/ckb-next/files/3304709/ckb-next-report-1560925757.gz)

## What is the program's current behavior?

When I boot my laptop and log in to Gnome (Ubuntu 18.04.2 LTS), every mouseclick left or right freezes the mouse completely for 1-2 seconds. Restarting the ckb-daemon fixes it until next reboot.

## What is the expected behavior of the program?

Not freezing and not requiring a daemon restart on every reboot.

## Any other notes & comments?

This _may_ have started with a kernel upgrade from 4.15.0-51 to 4.15.0-52. I have rebuilt and reinstalled ckb after the kernel upgrade and the problem appearing to see if that fixed would fix it.

tatokis, 2019-06-19:
It couldn't have been between -51 and -52.
https://kernel.ubuntu.com/git/kernel-ppa/mirror/ubuntu-bionic.git/log/?h=Ubuntu-4.15.0-52.56

When you say the mouse freezes, if there are any animations, do they also freeze?
Does the freeze happen if the GUI isn't running at all?

taisph, 2019-06-24:
Animations does not freeze with the mouse. It happens regardless of the GUI running until the daemon is restarted, although the animations only play when the GUI is running.

tatokis, 2019-06-24:
Would it be possible for you to run perf to see what calls take up a lot of time in the daemon?
Be sure to compile it with -DCMAKE_BUILD_TYPE=Debug

tatokis, 2019-06-25:
Another (easier) thing to try. Unbind the left click mouse button, and see if you can cause a freeze with it unbound. (GUI will need to be running.)

taisph, 2019-07-03:
I'm struggling with this one. I've been unable to reliably reproduce this and then just now while working, the mouse suddenly has high lag, updating only maybe twice a second or so. Animations are stalled and colors do not change on the mouse at all. Laptop is not doing any real work and there's no iowait. Stopping the daemon makes the mouse return to normal.

Compiled it with build type debug but it doesn't look like it produces much information unless I'm not looking in the right place.

Unbinding left mouse doesn't seem to have any effect.

The mouse lag varies with chosen animation profile and deleting the animation removes the lag completely which suggests that the lag is related to sending data to the mouse.

In the end I unplugged and plugged the mouse again and the problem disappeared which suggests buggy firmware or that something sent to the mouse can make it malfunction. Mouse firmware is up to date according to the CKB UI.

tatokis, 2019-08-16:
Any luck with this? I ran out of ideas...

taisph, 2019-08-28:
I've been running it on kernel 5.0.0-25-generic for a few days now. I did not recompile or reinstall ckb since the last test. So far I've not experienced any freezes or lag.

tatokis, 2019-08-29:
Glad to hear it. 
Feel free to close this issue when you are satisfied that the bug no longer happens.

taisph, 2019-09-06:
Still haven't had this issue after upgrading the kernel so let's hope it stays that way. Thanks.

closed 2019-09-06 by taisph