ckb-next/ckb-next #436: Script could not be loaded

patrickmay, 2019-06-22:
My Strafe keyboard on my iMac frequently reverts to all white lights.  When I open ckb-next, I get the error 'The "Wave" script could not be loaded.  Please check your animation directory.'  The only way I've found to fix it is to completely reinstall ckb-next, but then it happens again.

tatokis, 2019-06-23:
Please click the generate report button (in the settings tab) once this happens and attach the report to the issue

sjthespian, 2020-05-02:
I just started seeing the same thing this morning after a reboot, the report is attached.
[ckb-next-report-1588452969.gz](https://github.com/ckb-next/ckb-next/files/4568919/ckb-next-report-1588452969.gz)

closed 2022-10-03 by tatokis