ckb-next/ckb-next #437: Corsair Glaive RGB Pro not recognizing in ckb-next

blacksabbath9622, 2019-06-24:
Corsair Glaive RGB Pro not recognizing in ckb-next. I have tried all the troubleshooting methods as described on the main page. 

The mouse simply does not get detected in the application under the "ckb-next 0.4.0-27-g10e6b54" header and separator. I have attempted restarting my pc, and plugging the device in before/after/during BIOS load, and before/after/during ckb-next load after the operating system has been loaded, and the application automatically starts. I have tried switching USB ports, with/without the KVM that I use on this machine, and to no avail. So, I am submitting this issue followup request to see if anyone else has any ideas. Thanks.

I can attach the ckb-next generated log file upon request.

ZirconiumX, 2019-06-24:
You have a new device, so naturally the daemon won't detect it; we use a device whitelist. You will need to give us the generated report as the issue template requests.

blacksabbath9622, 2019-06-24:
I believe that I understand the issue. What I purchased was a Corsair Glaive RGB Pro, whereas the supported device is the non-Pro version of the Corsair Glaive. I apologize for making this mistake. Thank you for clearing up the confusion so quickly!

tatokis, 2019-06-25:
We can't tell if we can add support for your device without giving us a report. There's a reason we request it in the issue template.

Otherwise, you are welcome to add support for it yourself.

closed 2019-06-28 by tatokis