ckb-next/ckb-next #438: Leds not showing right color on profile switch on K95 Platinum

Termanater13, 2019-06-30:
The issue is when I switch profiles on my keyboard via the software or by the profiles button. The issue does get fixed when I mess with the color settings for that profile, or I adjust the brightness. I have tried to recreate the issue in windows via ICUE software and I could not, this leads me to think its ckb-next as the issue. This issue may also just be a hardware issue unique to my keyboard. Current color of the leds also do not seem to have an effect on the issue since I have 2 profiles when I switch back and forth the leds should not change color and the issue pops up, even when the profile key is used with one profile is after the other.

LEDs that are the issue are TOPBAR 4, 5, 15, 16, and 17.
Camera settings were altered to make the issue more apparent since default point and click on my Samsung S8 made it hard to see it when it is quite obvious to the eye. while the left and right are in separate pictures the issue does happen at the same time.

Issue on left side
[image]
left Side after adjusting brightness
[image]
Right Side Issue
[image]
Right Side after adjusting brightness
[image]

ckb-next version: 0.4.0-27-g10e6b54
k95 firmware: 3.21
Linux OS: Ubuntu 18.04

tatokis, 2019-08-16:
It's most likely a hardware issue. iCUE uses hardware brightness control, whereas ckb-next does it in software.
Try setting the brightness in iCUE all the way up and then when selecting colour, make it darker than usual, to simulate a "lower brightness".

tatokis, 2019-08-16:
Sorry, I completely misread your issue. It looks like an issue in the GUI possibly not detecting the colour changes.

Does this issue happen if you have an animation set?

Termanater13, 2019-08-18:
> Does this issue happen if you have an animation set?
This happens when the profiles changes, and animations fix the colors when the animation first activates. after the animation activates the issue is gone. If I did not have an animation set I would see the issue all the time.

wansti, 2020-05-19:
I'm having the exact same issue. Cycling brightness settings fixes it, switching to another profile brings it back.

Creative-user-name, 2021-07-14:
You might have done this by now, but for anybody new reading this (as I mentioned in #718) I "solved" it by animating the top bar in the same colour as the static colour and even forgot, that that was a thing.

This should also work, if you have different colours through the top bar, just hit it with a breathing animation, you could even make any animation random animation and just set its opacity to 0%

closed 2021-07-15 by tatokis

tatokis, 2021-07-15:
Some years and some frustration later, this should be fixed in master. It'd be appreciated if someone could verify it.

commit 941ff7dc (2021-07-15): Fix K95 Platinum topbar

Some LEDs would get overwritten as they accidentally shared
scancodes with the mouse dpi LEDs.

Closes #438