ckb-next/ckb-next #440: Device Support Request - M55 RGB PRO

Unaccounted4, 2019-07-03:
Report:
[ckb-next-report-1562169446.gz](https://github.com/ckb-next/ckb-next/files/3355768/ckb-next-report-1562169446.gz)

Corsair M55 RGB PRO 

Keep on rocking!

tatokis, 2019-07-03:
Unfortunately, the bad news is that despite this being a wired mouse, Corsair are using the new protocol that we don't have support for yet. It will probably be a while before we get it working. I think they'll be switching most new devices to it going forward, so we'll eventually have to support it.

That said, can you please run `sudo usbhid-dump -d 1b1c:1b70`

And after that, compile this https://github.com/ckb-next/getUsbString and run

```
sudo ./getUsbString 0x1b1c 0x1b70 4
```
and
```
sudo ./getUsbString 0x1b1c 0x1b70 5
```

Please paste the output of all of those here.

Unaccounted4, 2019-07-03:
Here you go: 
`sudo usbhid-dump -d 1b1c:1b70
003:012:003:DESCRIPTOR         1562186967.142327
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 97 15 00 25 01 75 01
 95 98 81 02 05 08 19 01 29 03 25 01 75 01 95 03
 91 02 75 05 95 01 91 01 C0

003:012:002:DESCRIPTOR         1562186967.142570
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

003:012:001:DESCRIPTOR         1562186967.142859
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

003:012:000:DESCRIPTOR         1562186967.143125
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 20
 15 00 25 01 95 20 75 01 81 02 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 75 08 95 01
 05 01 09 38 15 81 25 7F 81 06 C0 C0`

`sudo ./getUsbString 0x1b1c 0x1b70 4
Read 12 bytes
Raw (hex): 0c 03 56 00 30 00 33 00 30 00 30 00 
String: V0300`

`sudo ./getUsbString 0x1b1c 0x1b70 5
Read 10 bytes
Raw (hex): 0a 03 42 00 50 00 30 00 30 00 
String: BP00`

Halamix2, 2020-01-02:
What could be done to help getting some kind of support?

Wrizz, 2020-03-11:
Any updates on this? Would be greatly appreciated!

Halamix2, 2020-03-14:
I was able to decode only what is in gist (mode, DPI, color), I don't know how to get x/y data (so basically I can change color and DPI but cannot use the mouse as a mouse) or how to integrate any of that into ckb-next

Halamix2, 2020-03-18:
There is now incomplete and experimental Wireshark Dissector for Bragi: https://github.com/Halamix2/bragi_logs

KatanaKa-bedrock, 2021-04-14:
I tried what @Unaccounted4 did, but it sadly didn't help.

LuPro, 2021-05-04:
After seeing a thread about the new Ironclaw (#558) which also uses the bragi protocol and apparently had some preliminary test code for the bragi protocol added which allowed basic mouse functionality, I tried a lucky test if the bragi branch (or more like: the k95-xt branch, as for some reason git told me they are equal and forwarded me to that one?) would work for the M55; It did not :smile: 
I have 0 knowledge of the code base but stumbled my way around and did this:
- add a new macro for the device in `usb.h` (a macro `P_M55_RGB_PRO` with fitting product ID, as well as `IS_M55(kb)`)
- add device to `USES_BRAGI` macro in `usb.h`
- add said macro to `device_desc models[]` in `usb.c`
- add said macro to `product_str()` in `usb.c`

After compiling and running that my keyboard (K95 on old protocol) continued working, but the mouse broke - cursor was stuck and LEDs continued with their aggravating slowly-shifting rainbow :joy: 

Didn't really expect it to work as it's unfinished support for a new protocol made for a different device, but thought it was worth a shot - if you need anything I can help with (data, packets, testing, ...), please do tell, I'd love to support the endeavor of getting bragi properly supported (I'd love to (help) work on reverse engineering it myself and writing the code, but I have (almost) no prior experience with WireShark and frankly, too little time for a project of such a caliber).

andrey-shigantsov, 2021-06-25:
I need this. Is implementation expected?

GhostDog98, 2021-08-19:
Thought it might be useful to add some more data.
sudo usbhid-dump -d 1b1c:1b70
```
003:002:003:DESCRIPTOR         1629331488.953210
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 97 15 00 25 01 75 01
 95 98 81 02 05 08 19 01 29 03 25 01 75 01 95 03
 91 02 75 05 95 01 91 01 C0

003:002:002:DESCRIPTOR         1629331488.956204
 06 42 FF 09 02 A1 01 15 00 26 FF 00 75 08 95 40
 09 02 81 02 C0

003:002:001:DESCRIPTOR         1629331488.959208
 06 42 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 40
 09 01 81 02 95 40 09 01 91 02 C0

003:002:000:DESCRIPTOR         1629331488.962211
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 20
 15 00 25 01 95 20 75 01 81 02 75 10 95 02 05 01
 09 30 09 31 16 01 80 26 FF 7F 81 06 75 08 95 01
 05 01 09 38 15 81 25 7F 81 06 C0 C0
```
And 
```
sudo ./getUsbString 0x1b1c 0x1b70 4
Read 12 bytes
Raw (hex): 0c 03 56 00 30 00 33 00 30 00 30 00 
String: V0300
```
```
sudo ./getUsbString 0x1b1c 0x1b70 5
Read 10 bytes
Raw (hex): 0a 03 42 00 50 00 30 00 30 00 
String: BP00
```
I might boot back into windows later and programmatically wireshark a bunch of options for the mouse. Will update if i get time.

agoshigagoroba, 2021-12-31:
bump to this

noneistaken, 2022-02-06:
Bump, I have this mouse too.

closed 2022-02-20 by tatokis

n0n467, 2022-04-10:
The M55 mouse may be supported but I have an M55 RGB PRO and it's not detected.

n0n467, 2022-04-10:
@Halamix2 I don't really know much about these things but would it be easy to add support for the M55 RGB PRO or is it to different from the M55 ?

tatokis, 2022-04-10:
@n0n467 You probably need to update ckb-next. You have to run a ckb-next-git package (or manually build from git). Support for the mouse was added after 0.4.4 was released.

n0n467, 2022-04-10:
@tatokis I already built the latest version from source but my mouse still isn't detected.

tatokis, 2022-04-10:
If it's wireless, it's a different device and not yet supported. If not, generate a report and attach it.

n0n467, 2022-04-10:
@tatokis how can I make a report ?

n0n467, 2022-04-10:
Sorry, after uninstalling and reinstalling the same version, it started working.
Thank for your fast replies anyway.