ckb-next/ckb-next #441: fn keys combinations

patvdleer, 2019-07-08:
Is it possible to set F13 on Fn+F1, I don't know if the keyboard is able to send this itself (like a laptop does) or it  needs to be handled by software.

## What is the program's current behaviour?
Select fn and F1 (with control) sets  both keys to F13

## What would you like it to do instead?
Select fn and F1 (with control) set the combo to F13

## Any other notes & comments?
https://github.com/ckb-next/ckb-next/issues/120#issuecomment-508581728

tatokis, 2019-08-16:
As far as I know, the keyboard itself can only send Fn + whatever is defined in the number keys. I don't think the keyboard sends a scancode for Fn itself.

If it does, it might be possible to implement it, however the code is currently not written to support actions on multiple keys pressed at once.

Manyy8, 2026-11-10:
Hi! I'd be interested in contributing to this issue.

I understand that the main challenge is handling simultaneous key combinations and determining whether the Fn key can be detected independently by the software.
Is this issue still open for contributions? Also, has there been any progress or investigation into the hardware limitations mentioned in the discussion?

I'd be happy to investigate the existing implementation and explore a possible solution.