ckb-next/ckb-next #443: Glaive RGB Pro Support

themikem, 2019-07-14:
# Before you proceed

- Check if your device is listed in the [wiki](https://github.com/ckb-next/ckb-next/wiki/Supported-Hardware).
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

**Click the `Generate Report` button in the `Settings` tab of the ckb-next user interface and attach the generated report to this issue.**

---
Looking for Glaive RGB Pro support.  The only difference I see with the "non-pro" Glaive is the addition of a second button to change DPI / speed profile.  Otherwise seems identical.

Report attached.
[ckb-next-report-1563131175.gz](https://github.com/ckb-next/ckb-next/files/3390345/ckb-next-report-1563131175.gz)

Thanks!

tatokis, 2019-08-16:
Looks like it uses the old NXP protocol (which we support).

The code in the `glaive-rgb-pro` branch should make it show up as a Glaive RGB.
Try it tell us what works and what doesn't.

https://github.com/ckb-next/ckb-next/tree/glaive-rgb-pro
https://github.com/ckb-next/ckb-next/commit/26c7b62dde24a8d39cf4ccaa698d0c136b22d1db

themikem, 2019-08-19:
Awesome.  Thanks for doing this.

I checked out that branch and ran ./quickinstall.  Mouse no longer works after running this, and it sends random keystrokes to the terminal like a keyboard, including enter, so it tries to run commands.

On the flip side, it managed to push up arrow and enter somehow, so it seems capable of rebuilding itself.  Now if we can just get that automatic bug fix feature working..... ("Please move your mouse in random circles until all bugs are fixed....")

drsxr, 2019-08-20:
Hey - have a Glaive RGB so just wanted to express my thanks to the maintainers and authors of this distro.

tatokis, 2019-08-20:
@themikem I forgot to add it as a mouse and it was being recognised as a keyboard.
Can you please try again with the latest code?

themikem, 2019-09-15:
@tatokis First, thanks for looking at this.  Much appreciated!

The fix worked!  Mouse is functional, including the RGB lighting, a hardware profile, and proper detection of DPI mode change using both up and down DPI mode buttons.  Didn't extensively test the "save to hardware" feature, but it does not cause any issues when you push the button in the UI.

Only issues I see:
- the "bar graph" DPI level indicator does not light up at all.  Is there anything I can do to test this further than changing colors of the various DPI levels?
- Binding page will need a "DPI down" button as well - that's basically the only hardware difference I see on this mouse, outside of the restyled modular grip thing.

Let me know if there is any further data that would help.

Thanks again.

DaiKaiser93, 2020-07-23:
Does anyone know if it's going to get added to the master branch? Mine works fine but I'm unable to customize the RGB as it's not detected by CKB

Ravenslofty, 2020-07-23:
We'll probably need some packet traces from CUE to figure out what's happening with the DPI indicator LEDs. That means you'll need to install Wireshark on Windows, open CUE up, and press the DPI up/down buttons a few times. From there we might be able to work out what changed.

nickvde, 2021-05-31:
Is there any progress yet? I can help possibly with wireshark captures.

Runlevel-zero, 2021-12-04:
I was going to submit a report but I am assuming this is still an issue and preventing the glaive RGB pro from being detected?

tatokis, 2021-12-04:
First of all, if the device isn't exactly mentioned in the wiki, it's not supported. It says "Glaive", not "Glaive pro".
Secondly, we requested packet traces and no one provided them.

I can rebase the glaive-rgb-pro branch to bring it up to date, if you want to get the functionality mentioned above and help get it fully supported, but unless someone invests time to help us, we can't really do much.

tatokis, 2021-12-04:
I have rebased the code and added the dpi down button as well. Someone please test the `glaive-pro-rgb-v3` branch and test the dpi down button.

Runlevel-zero, 2021-12-05:
@tatokis Thanks for the response. Sorry that came off more accusatory than I intended, I'm sure the wiki entry predates the pro, I get it. I just wanted to explain why I was confused.

Anyhow, 

The glaive-pro-rgb-v3 branch properly binds the dpi down button and it con be rebound to other functions. However, there is no dpi indication on the mouse itself.

tatokis, 2021-12-10:
@Runlevel-zero Apologies, didn't mean to sound harsh. I was describing what the situation was.

While working on something else, I saw that an alternative DPI indicator implementation was implemented for the Dark Core by someone else. It's possible that the same applies to the Glaive RGB Pro.

I have added that for this mouse as well. Can you please test the latest code from the same branch?

DaiKaiser93, 2021-12-11:
@tatokis The DPI indicator is working now, with full RGB control

closed 2022-04-04 by tatokis