ckb-next/ckb-next #444: Crsair Veangeance RGB Pro

s7relok, 2019-07-23:
Hello,

I will soon have a pack of Corsair Veangeance RGB Pro RAM sticks.

Can you tell me what type of info you will need to help to build the support? 

Regards,

michaelothing, 2019-08-18:
If you dual boot this ram will keep it's settings if set from windows icue unlike the rest of the products. I'm looking for a solution for the AIO water coolers.

kmasaryk, 2019-08-22:
@michaelothing You're assuming people have Windows installed. I do not, nor do I have any plans to do so. It would be great to get this working with the Vengeance RGB sticks.

If your using an ASUS mb, it sounds like [OpenAuraSDK](https://gitlab.com/CalcProgrammer1/OpenAuraSDK) might work.

michaelothing, 2019-08-22:
I assume nothing. I simply stated a fact that it holds settings from icue. I have signed in to windows once in the last year.

gravityfargo, 2023-04-21:
closing in favor of #589

closed 2023-04-21 by gravityfargo