ckb-next/ckb-next #445: Failed to install

tad1073, 2019-07-24:
- Search in open and closed [Issues](https://github.com/ckb-next/ckb-next/issues).
- If possible, try the latest code by building the software manually.
- File unrelated issues separately.

## General information
Failed to install
### Source
How and where did you get this program, e.g.:
  * Package from AUR/PPA/COPR/...
  * Built manually using quickinstall script
  * Pre-built dmg for macOS
./quickinstall

## Logs & crash reports
[CMakeError.log](https://github.com/ckb-next/ckb-next/files/3426251/CMakeError.log)
[CMakeOutput.log](https://github.com/ckb-next/ckb-next/files/3426252/CMakeOutput.log)

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?

## What is the expected behaviour of the program?

## Any other notes & comments?

ZirconiumX, 2019-07-24:
I can't even tell what the problem is from your logs. They're completely unhelpful.

tatokis, 2019-07-24:
There doesn't seem to be an error based on your logs. The iconv "error" is normal, as it compiles a test program to test what features are available. It's normal for that program to fail on linux, so cmake marks it accordingly.

tad1073, 2019-07-24:
[~/.tmp/ckb-next]: sudo ./quickinstall
[sudo] password for thomas: 
-- The C compiler identification is GNU 7.4.0
-- The CXX compiler identification is GNU 7.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- ckb-next version: 0.4.0-34-g5621e20 (Non-release)
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST
-- Performing Test ICONV_SECOND_ARGUMENT_IS_CONST - Failed
-- Found Iconv: /usr/lib/x86_64-linux-gnu/libc.so
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Found UDEV: /lib/x86_64-linux-gnu/libudev.so  
-- Searching for running ckb and/or ckb-next GUI
-- NOTE:
    Privileged access is required for operations upon the daemon at
    configure time as it is owned by root. If this is unacceptable,
    disable SAFE_INSTALL and prepare the filesystem manually.
-- systemd detected
-- Generating and importing ckb-next-daemon.service (systemd)
inactive
Failed to get unit file state for ckb-daemon.service: No such file or directory
inactive
Failed to get unit file state for ckb-next-daemon.service: No such file or directory
-- Found QuaZip: /usr/lib/x86_64-linux-gnu/libquazip5.so  
-- Found PulseAudioSimple: /usr/lib/x86_64-linux-gnu/libpulse-simple.so  
-- Found PulseAudio: /usr/lib/x86_64-linux-gnu/libpulse.so  
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
  The imported target "Qt5::Gui" references the file

     "/usr/lib/x86_64-linux-gnu/libEGL.so"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:50 (_qt5_Gui_check_file_exists)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:72 (_qt5gui_find_extra_libs)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:184 (include)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:101 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  src/gui/CMakeLists.txt:72 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/thomas/.tmp/ckb-next/build/CMakeFiles/CMakeOutput.log".
See also "/home/thomas/.tmp/ckb-next/build/CMakeFiles/CMakeError.log".
An error occurred,
press enter to exit.

ZirconiumX, 2019-07-24:
You need to install EGL, it seems. This isn't a bug on our end.

tad1073, 2019-07-24:
Thanks, after I re-read the output I realized what the problem was.

closed 2019-07-24 by ZirconiumX

tatokis, 2019-07-24:
I'd like to add that If you're using nvidia graphics, it most likely happened because you installed the driver using their .run tool and not the graphics ppa https://launchpad.net/~graphics-drivers/+archive/ubuntu/ppa

If you're lucky enough, uninstalling the .run driver and installing the one from the ppa should easily fix this.
Either way, as @ZirconiumX said this isn't a bug on our end

tad1073, 2019-07-24:
I'm still usable to install ckb. I'm missing some other dependencies but
don't know which ones.

tatokis, 2019-07-24:
https://github.com/ckb-next/ckb-next/wiki/Linux-Installation#build-from-source

tad1073, 2019-07-24:
I've followed the tut, I have all those dependencies installed. It has
something to do with the qt kde libraries because I installed kde desktop
then uninstalled it using synaptic package manager. How hard would a gtk
version need to code?