ckb-next/ckb-next #447: Best way to permantly change keyboard RGB color?

GideonWolfe, 2019-07-27:
So basically I am writing a script that themes all aspects of my system based on colors extracted from an image. From these colors, I pick a bright one and pipe it into the following command:

`echo $KEYCOLOR > /dev/input/ckb1/cmd`

This will change the keycolor, with a few caveats. It doesn't work when there is an animation applied, which I read about on another issue. I am only using a simple heatmap, would there be a way to reapply the animation after the color change?

Also, the color doesn't seem to stick, because if I try to increase the brightness, it snaps back to the color I set in the GUI.

What would be the best way to script the permanent color change? The animation would be a bonus, but I really just need the color to stick.

tatokis, 2019-09-01:
https://github.com/ckb-next/ckb-next/wiki/Animations#pipe
This should get you going

closed 2019-09-01 by tatokis

GideonWolfe, 2019-09-01:
Just to clarify: 
I opened CKB-next and enabled the "Pipe" animation for my keyboard.
I extract my desired color with `KEYBOARDCOLOR=$(cat $HOME/.cache/wal/colors.css | grep color6 | awk '{print $2}' | cut -c2- | rev | cut -c2- | rev)`

I then run `echo $KEYBOARDCOLOR > /tmp/ckbpipe000`, but nothing happens.

I also tried the examples like `echo "rgb ff00ffff" > /tmp/ckbpipe000` with the same result. It says the command without errors, but the keyboard color isn't changing.

Is there a stem i'm missing?