ckb-next/ckb-next #450: Cannot write to hardware, settings get lost as soon as ckb-next application is quit

sumitkm, 2019-07-31:
## General information
I am trying to setup a Corsair Strafe RGB (not MK2) for the first time using ckb-next and I can't seem to save any of the profiles to hardware. Maybe I am getting this wrong, but isn't "Save to hardware" suppose to update the keyboard hardware so I can plug it out and attach it to some other machine and still have the theme available in the new machine?

### Source
- Installed latest dmg from the releases folder (0.4.x) on OSX
- Built it on Debian Linux
Both have the same behaviour.

## What is the program's current behaviour?
When I say save to hardware all the keys on the keyboard blink once (as if something happend) but as soon as I close ckb-next the profile reverts to the default 'red' keys if I choose HWProfile and White if I chose Demo profile.

## What is the expected behaviour of the program?
I am expecting the profile I change using ckb-next to persist without ckb-next running (for example I set a the ripple scheme on OSX) I should be able to unplug and plug into XBox and see the same effect

## Any other notes & comments?
I don't have a Windows computer at home so I have never plugged this into an Windows machine.

ZirconiumX, 2019-08-01:
Corsair keyboards prior to the K95 Platinum and MK.2 series - like your Strafe RGB - as well as some old and/or low-end mice only support saving a single static colour per lighting zone to hardware.

The newer keyboards have full animation support, but because (the original) ckb was not designed for hardware animations, and I haven't done much reverse engineering of the protocol or code redesign, because I'm a little tired of working on ckb.

closed 2019-08-01 by ZirconiumX

sumitkm, 2019-08-01:
Thank you for the update. Also many thanks for all the work so far. 

> Corsair keyboards prior to the K95 Platinum and MK.2 series - like your Strafe RGB - as well as some old and/or low-end mice only support saving a single static colour per lighting zone to hardware.

Just a quick clarification if you have a moment, is it ckb's limitation (lack of protocol reverse engineering time or information available) or the (older) keyboard hardware's limitation that we can't save animations to the KB?

tatokis, 2019-08-16:
Your Strafe RGB does not support saving animations. You can only save the colour on the keys.

The MK2 does support saving animations, but ckb-next also doesn't have support for saving animations to it.