ckb-next/ckb-next #451: CKB Not Launching at Start-Up

jgonzales57, 2019-08-01:
Hello,
I have been using CKB for about a year now, and it was working perfectly fine.  I upgraded to a new monitor and had to adjust some display configure files to set the correct resolution etc.  However, now CKB won't launch when the computer is started up, and when I open it manually, the window is MASSIVE, it takes multiple drags across the entire display to go from one end of the window to the other, any suggestions?

tatokis, 2019-08-15:
Please try the latest code from git/master.
It should be fixed by https://github.com/ckb-next/ckb-next/commit/59c070cbc0f3f580315d83dcdbaf9291d6afe15f

jgonzales57, 2019-08-16:
Reinstalled using the newest version and it is working great!  Thanks for your help!

closed 2019-08-16 by jgonzales57