ckb-next/ckb-next #455: GUI prevents Plasma 5 from logging out

tatokis, 2019-08-23:
If the GUI is running and displaying its MainWindow, (minimized as well), attempting to shut down or log off will result in the GUI canceling the action.
`Logout canceled by '/usr/local/bin/ckb-next'`

This does not seem to be an issue when it's hidden in the tray, so possibly the issue is somewhere in the "Hit X to hide to tray" code.

WildPenquin, 2019-10-21:
I'm having this issue even when ckb-next is minimized to the tray,

Grymas, 2020-05-04:
Same issue here. 
Since i'm forcing it to the tray on startup (because it doesn't want to for some reason) with kde window rules, this is rather annoying. It looks in the tray for me and prevents logout/shutdown the first time I try.

tatokis, 2020-05-04:
It's impossible to fix, because KDE seems to act as if it pressed the X button in it, and we have the X button set to minimize the GUI to tray.

I added a workaround in git master to make it start in tray in KDE, so this should not be an issue.
https://github.com/ckb-next/ckb-next/commit/4bf32bd51281904c8b787b7b325d65ff90125af5