ckb-next/ckb-next #456: Keyboard does not respond on wake from sleep

greenfox1505, 2019-08-25:
## General information

### Source
  * Built manually using quickinstall script

## What is the program's current behaviour?

When I wake my tower from sleep, the Corsair keyboard locks-up and does not respond. Occasionally this lock-up occurs during normal use. Other keyboards plugged into the machine (either connected already or added after the lock-up) respond normally.

The lock-up can be solved by either unplugging/replugging the keyboard or by running `service ckb-next-daemon restart`.

## What is the expected behaviour of the program?

Keyboard should just work.

tatokis, 2019-08-28:
Most of the changes in the update you already had from master, so it probably doesn't make too much of a difference.

That said, after resuming from sleep (after fixing the keyboard), can you post the output of `dmesg | grep usbfs` and also generate and attach a report

tatokis, 2019-08-28:
Please try the code from the `reclaim-ifs` branch.
https://github.com/ckb-next/ckb-next/commit/7782fd80305b4d9b885471fc957e72a21f84e0be

greenfox1505, 2019-09-01:
Running `ckb-next 0.4.1-6-g7782fd8`

```
[59917.937494] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[59918.044335] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[62368.569579] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[62368.643588] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[64789.439747] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[64789.587734] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[68424.102638] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[68424.194135] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[70872.355741] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[70872.446920] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[73296.358937] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[73296.502425] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[75718.293627] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[75718.378812] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
[78167.886595] usb 3-10.1: usbfs: process 29601 (suspend) did not claim interface 1 before use
[78167.969309] usb 3-10.1: usbfs: process 29615 (ckb1 usb) did not claim interface 1 before use
```

(obviously this has happened a few times since my last boot)

There are also some `input: Corsair CORSAIR K70 RGB MK.2` lines here. Should I include them next time?

tatokis, 2019-09-05:
You need to run the code from the `reclaim-ifs` branch. If you are using git, you can do `git checkout reclaim-ifs`, otherwise, download the zip archive from here https://github.com/ckb-next/ckb-next/tree/reclaim-ifs

greenfox1505, 2019-09-07:
I did that. Then I ran `./quickinstall`. Does that not upgrade the install? I did it again after a `git pull` and the version number did not change.

tatokis, 2019-09-08:
Apologies. Seems to be a mistake on my end. Looks like this doesn't resolve the issue.
That said, I'm not sure the using without claiming warning is directly related either.

greenfox1505, 2019-09-09:
I added this script and it seems to function as a workaround

```bash
#!/bin/sh
#/usr/lib/systemd/system-sleep/keyboardDriverRestart.sh

case $1 in
  post)
	sudo service ckb-next-daemon restart
    ;;
esac
```

it works. it's ugly, but it works.

Momper14, 2021-01-01:
Had a similar problem. After sleep my Keyboard lights up but didn't responded to key presses but this fixed it for me too. TY :)

maxlemieux, 2021-04-11:
Me too. Lots of spam in dmesg like this:
```
[215004.706172] usb 1-7: usbfs: process 1795 (ckb1 input) did not claim interface 0 before use
[215004.712323] usb 1-8: usbfs: process 1811 (ckb2 input) did not claim interface 0 before use
[215004.717787] usb 1-7: usbfs: process 1397 (suspend) did not claim interface 1 before use
[215004.817688] usb 1-8: usbfs: process 1397 (suspend) did not claim interface 1 before use
[215005.336634] usb 1-8: usbfs: process 1811 (ckb2 input) did not claim interface 0 before use
[223139.108820] usb 1-7: usbfs: process 1795 (ckb1 input) did not claim interface 0 before use
[223139.108876] usb 1-8: usbfs: process 1811 (ckb2 input) did not claim interface 0 before use
[223139.113880] usb 1-7: usbfs: process 1397 (suspend) did not claim interface 1 before use
[223139.227127] usb 1-8: usbfs: process 1397 (suspend) did not claim interface 1 before use
```
Adding greenfox1505's script fixed it for me, thanks. I wonder if there's any downside to making this part of the default installation for ckb-next, on environments which use systemd?

Distro: Ubuntu 21.04, kernel: 5.11.0-13-generic

BloodyIron, 2021-06-16:
I'm getting this for my mouse on Ubuntu 20.04, have to replug it. I'll try to restart the daemon next time instead, but this really should get fixed in the daemon itself.

BloodyIron, 2021-06-17:
Okay so restarting the daemon, the mouse comes back, but loads a completely default configuration until I unplug and replug the mouse back into the computer. NOT a solution!

bobbruno, 2021-06-20:
That script didn't work for me either. Even adding a "pre) sudo service ckb-next-daemon stop" to the script didn't help. My usb goes into some locked state and nothing other than a hard reset releases. Even a "sudo shutdown" won't work, I get a number of messages about USB not being released or something like this.

ion201, 2022-03-07:
This appears to be resolved in the master branch - I haven't seen this problem since rebuilding a few days ago.

ollpu, 2022-03-07:
Yes, should be more or less fixed by my PR. However, I did experience some issues in case a suspend is attempted but fails for some (unrelated) reason. I should probably file a separate issue for that.

BloodyIron, 2022-03-08:
Can't wait for my mouse to stop turning the lights off when resuming from suspend lol!

gravityfargo, 2024-05-01:
This issue has been marked as stale due to age and/or lack of activity. 

Possibly fixed with #777

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-01 by gravityfargo