ckb-next/ckb-next #459: No Allow button in system preferences- security and privacy

neeldug, 2019-08-27:
## General information

### Source
How and where did you get this program, e.g.:
Pre-built dmg for macOS

## What is the program's current behaviour?
After installing states: Warning: System extension not allowed, have installed three times, manually as well. Have restarted, unplugged devices, still to no avail. Also, have been able to click allow to other extensions.

## What is the expected behaviour of the program?
Allow button should be visible in security and privacy tab, however, is not.
## Any other notes & comments?

tatokis, 2019-08-27:
Please generate and attach a report

neeldug, 2019-09-02:
```
2019; root:xnu-4903.270.47~4/RELEASE_X86_64 x86_64

OS Version:
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G87

USB Devices:
        Corsair STRAFE RGB Gaming Keyboard:

          Product ID: 0x1b20
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 3.08
          Serial Number: 0F00D02DAEAA100455029C83F5001944
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 13
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

--
--
        Corsair VOID RGB USB Gaming Headset :

          Product ID: 0x1b2a
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.02
          Serial Number: 00000000
          Speed: Up to 12 Mb/sec
          Manufacturer: Cosair
          Location ID: 0x14200000 / 6
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

IOReg:
+-o Corsair VOID RGB USB Gaming Headset @14200000  <class AppleUSBDevice, id 0x100000355, registered, matched, active, busy 0 (5 ms), retain 20>
  | {
  |   "sessionID" = 0x6bfa8d6b
  |   "iManufacturer" = 0x1
  |   "bNumConfigurations" = 0x1
  |   "idProduct" = 0x1b2a
  |   "bcdDevice" = 0x2
  |   "Bus Power Available" = 0xfa
  |   "USB Address" = 0x6
  |   "bMaxPacketSize0" = 0x40
  |   "iProduct" = 0x2
  |   "iSerialNumber" = 0x3
  |   "bDeviceClass" = 0x0
  |   "Built-In" = No
  |   "locationID" = 0x14200000
  |   "bDeviceSubClass" = 0x0
  |   "bcdUSB" = 0x200
  |   "USB Product Name" = "Corsair VOID RGB USB Gaming Headset "
  |   "PortNum" = 0x2
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0x0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0x0,"CurrentPowerState"=0x3,"CapabilityFlags"=0x10000,"MaxPowerState"=0x4,"DriverPowerState"=0x3}
  |   "kUSBCurrentConfiguration" = 0x1
  |   "Device Speed" = 0x1
  |   "USB Vendor Name" = "Cosair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "00000000"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
[... 255 lines cut ...]
        Info.plist

[I] Waiting for kext...
[F] os_inputopen (input_mac.c:77): Unable to open VirtualHIDDevice
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1

GUI log:
No GPG2 or GPG detected. Signature verification disabled.
Using QSytemTrayIcon
Path to settings: "/Users/Neel/.config/ckb-next/ckb-next.ini"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/heat"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/invaders"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/life"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pinwheel"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pipe"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/rain"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/snake"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.

```

tatokis, 2019-09-05:
I have no idea how this happened, but if you run this in the terminal and then unplug/replug your keyboard (or possibly reboot too), it should show the allow button.

```
sudo chown -R root:wheel /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext && sudo chmod -R 755 /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext
```

neeldug, 2019-09-08:
Still no change, have rebooted after running command, and unplugged/replugged, error message appears to remain the same:

`OS Version:
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G95

USB Devices:
        Corsair STRAFE RGB Gaming Keyboard:

          Product ID: 0x1b20
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 3.08
          Serial Number: 0F00D02DAEAA100455029C83F5001944
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 12
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

--
--
        Corsair VOID RGB USB Gaming Headset :

          Product ID: 0x1b2a
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.02
          Serial Number: 00000000
          Speed: Up to 12 Mb/sec
          Manufacturer: Cosair
          Location ID: 0x14200000 / 11
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

IOReg:
+-o Corsair VOID RGB USB Gaming Headset @14200000  <class AppleUSBDevice, id 0x100000701, registered, matched, active, busy 0 (714 ms), retain 21>
  | {
  |   "sessionID" = 0x3ab9ec3cf7
  |   "iManufacturer" = 0x1
  |   "bNumConfigurations" = 0x1
  |   "idProduct" = 0x1b2a
  |   "bcdDevice" = 0x2
  |   "Bus Power Available" = 0xfa
  |   "USB Address" = 0xb
  |   "bMaxPacketSize0" = 0x40
  |   "iProduct" = 0x2
  |   "iSerialNumber" = 0x3
  |   "bDeviceClass" = 0x0
  |   "Built-In" = No
  |   "locationID" = 0x14200000
  |   "bDeviceSubClass" = 0x0
  |   "bcdUSB" = 0x200
  |   "USB Product Name" = "Corsair VOID RGB USB Gaming Headset "
  |   "PortNum" = 0x2
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0x0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0x0,"CurrentPowerState"=0x3,"CapabilityFlags"=0x10000,"MaxPowerState"=0x4,"DriverPowerState"=0x3}
  |   "kUSBCurrentConfiguration" = 0x1
  |   "Device Speed" = 0x1
  |   "USB Vendor Name" = "Cosair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "00000000"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
  | 
[... 252 lines cut ...]
        CodeResources
        VirtualHIDDevice
        Info.plist

[I] Waiting for kext...
[F] os_inputopen (input_mac.c:77): Unable to open VirtualHIDDevice
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1

GUI log:
No GPG2 or GPG detected. Signature verification disabled.
Using QSytemTrayIcon
Path to settings: "/Users/Neel/.config/ckb-next/ckb-next.ini"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/heat"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/invaders"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/life"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pinwheel"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pipe"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/rain"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/snake"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.`

tatokis, 2019-09-08:
What is the output of `ls -la /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext` ?

neeldug, 2019-09-09:
```
total 0
drwxr-xr-x   3 root  wheel   96 31 Jul  2018 .
drwxr-xr-x  27 root  wheel  864  5 Sep 18:52 ..
drwxr-xr-x   5 root  wheel  160  2 Sep 16:47 Contents
```

tatokis, 2019-09-13:
Can you uninstall ckb-next from the button in the settings tab, and then reinstall it and run 
`ls -la /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext/Contents` ?

tatokis, 2019-09-13:
Can you uninstall ckb-next from the button in the settings tab, then run
```
sudo rm -rf /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext/
```
and then reinstall it and run 
```
ls -la /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext/Contents
``` 
?

neeldug, 2019-09-16:
Here's the result: 

```
total 8
drwxr-xr-x  5 root  wheel   160  6 Jun 06:47 .
drwxr-xr-x  3 root  wheel    96  6 Jun 06:47 ..
-rwxr-xr-x  1 root  wheel  2339  6 Jun 06:47 Info.plist
drwxr-xr-x  3 root  wheel    96  6 Jun 06:47 MacOS
drwxr-xr-x  3 root  wheel    96  6 Jun 06:47 _CodeSignature
```

tatokis, 2019-09-16:
And you still get no allow button? Those look like the correct permissions...
Can you please check again in the privacy menu and possibly provide another report?

neeldug, 2019-09-17:
Still no allow button, but here's the current report. The cause of this was I had to restore my machine from a backup.

Here's the report anyway:

```
OS Version:
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G95

USB Devices:
        Corsair STRAFE RGB Gaming Keyboard:

          Product ID: 0x1b20
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 3.08
          Serial Number: 0F00D02DAEAA100455029C83F5001944
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 7
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

--
--
        Corsair VOID RGB USB Gaming Headset :

          Product ID: 0x1b2a
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.02
          Serial Number: 00000000
          Speed: Up to 12 Mb/sec
          Manufacturer: Cosair
          Location ID: 0x14200000 / 3
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

IOReg:
+-o Corsair VOID RGB USB Gaming Headset @14200000  <class AppleUSBDevice, id 0x100000321, registered, matched, active, busy 0 (5 ms), retain 20>
  | {
  |   "sessionID" = 0x57f17318
  |   "iManufacturer" = 0x1
  |   "bNumConfigurations" = 0x1
  |   "idProduct" = 0x1b2a
  |   "bcdDevice" = 0x2
  |   "Bus Power Available" = 0xfa
  |   "USB Address" = 0x3
  |   "bMaxPacketSize0" = 0x40
  |   "iProduct" = 0x2
  |   "iSerialNumber" = 0x3
  |   "bDeviceClass" = 0x0
  |   "Built-In" = No
  |   "locationID" = 0x14200000
  |   "bDeviceSubClass" = 0x0
  |   "bcdUSB" = 0x200
  |   "USB Product Name" = "Corsair VOID RGB USB Gaming Headset "
  |   "PortNum" = 0x2
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0x0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0x0,"CurrentPowerState"=0x3,"CapabilityFlags"=0x10000,"MaxPowerState"=0x4,"DriverPowerState"=0x3}
  |   "kUSBCurrentConfiguration" = 0x1
  |   "Device Speed" = 0x1
  |   "USB Vendor Name" = "Cosair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "00000000"
[... 262 lines cut ...]
        VirtualHIDDevice
        Info.plist

[I] Waiting for kext...
[F] os_inputopen (input_mac.c:77): Unable to open VirtualHIDDevice
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1

GUI log:
No GPG2 or GPG detected. Signature verification disabled.
Using QSytemTrayIcon
Path to settings: "/Users/Neel/.config/ckb-next/ckb-next.ini"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/heat"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/invaders"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/life"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pinwheel"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pipe"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/rain"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/snake"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.
```

tatokis, 2019-09-17:
```
Kext rejected due to improper filesystem permissions: <OSKext 0x7f8a4ae3a340 [0x7fffa87f08e0]> { URL = "file:///Library/StagedExtensions/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext/", ID = "org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800" }
Diagnostics for /Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext:
Authentication Failures: 
    File owner/permissions are incorrect (must be root:wheel, nonwritable by group/other): 
        CodeResources
        VirtualHIDDevice
        Info.plist
```
Yet the permissions are correct. I don't understand what the problem is.
```
-rwxr-xr-x  1 root  wheel  2339  6 Jun 06:47 Info.plist
```

Try running 
```
sudo kextcache --clear-staging
```
It might fix this, since the permissions are obviously correct.

neeldug, 2019-09-17:
Still no change, same error log:

```
OS Version:
ProductName:	Mac OS X
ProductVersion:	10.14.6
BuildVersion:	18G95

USB Devices:
        Corsair STRAFE RGB Gaming Keyboard:

          Product ID: 0x1b20
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 3.08
          Serial Number: 0F00D02DAEAA100455029C83F5001944
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14400000 / 7
          Current Available (mA): 500
          Current Required (mA): 500
          Extra Operating Current (mA): 0

--
--
        Corsair VOID RGB USB Gaming Headset :

          Product ID: 0x1b2a
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 0.02
          Serial Number: 00000000
          Speed: Up to 12 Mb/sec
          Manufacturer: Cosair
          Location ID: 0x14200000 / 3
          Current Available (mA): 500
          Current Required (mA): 100
          Extra Operating Current (mA): 0

IOReg:
+-o Corsair VOID RGB USB Gaming Headset @14200000  <class AppleUSBDevice, id 0x100000321, registered, matched, active, busy 0 (97 ms), retain 20>
  | {
  |   "sessionID" = 0x57f17318
  |   "iManufacturer" = 0x1
  |   "bNumConfigurations" = 0x1
  |   "idProduct" = 0x1b2a
  |   "bcdDevice" = 0x2
  |   "Bus Power Available" = 0xfa
  |   "USB Address" = 0x3
  |   "bMaxPacketSize0" = 0x40
  |   "iProduct" = 0x2
  |   "iSerialNumber" = 0x3
  |   "bDeviceClass" = 0x0
  |   "Built-In" = No
  |   "locationID" = 0x14200000
  |   "bDeviceSubClass" = 0x0
  |   "bcdUSB" = 0x200
  |   "USB Product Name" = "Corsair VOID RGB USB Gaming Headset "
  |   "PortNum" = 0x2
  |   "non-removable" = "no"
  |   "IOCFPlugInTypes" = {"9dc7b780-9ec0-11d4-a54f-000a27052861"="IOUSBFamily.kext/Contents/PlugIns/IOUSBLib.bundle"}
  |   "bDeviceProtocol" = 0x0
  |   "IOUserClientClass" = "IOUSBDeviceUserClientV2"
  |   "IOPowerManagement" = {"DevicePowerState"=0x0,"CurrentPowerState"=0x3,"CapabilityFlags"=0x10000,"MaxPowerState"=0x4,"DriverPowerState"=0x3}
  |   "kUSBCurrentConfiguration" = 0x1
  |   "Device Speed" = 0x1
  |   "USB Vendor Name" = "Cosair"
  |   "idVendor" = 0x1b1c
  |   "IOGeneralInterest" = "IOCommand is not serializable"
  |   "USB Serial Number" = "00000000"
  |   "IOClassNameOverride" = "IOUSBDevice"
  | }
[... 261 lines cut ...]
        Info.plist

[I] Waiting for kext...
[F] os_inputopen (input_mac.c:77): Unable to open VirtualHIDDevice
[I] Disconnecting /var/run/ckb1
[I] Removed device path /var/run/ckb1

GUI log:
No GPG2 or GPG detected. Signature verification disabled.
Using QSytemTrayIcon
Path to settings: "/Users/Neel/.config/ckb-next/ckb-next.ini"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/gradient"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/heat"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/invaders"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/life"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pinwheel"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/pipe"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/rain"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/random"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/ripple"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/snake"
Scanning "/Applications/ckb-next.app/Contents/Resources/ckb-next-animations/wave"
Downloaded new firmware list. 38 entries found.
objc[476]: Class FIFinderSyncExtensionHost is implemented in both /System/Library/PrivateFrameworks/FinderKit.framework/Versions/A/FinderKit (0x7fffac0903d8) and /System/Library/PrivateFrameworks/FileProvider.framework/OverrideBundles/FinderSyncCollaborationFileProviderOverride.bundle/Contents/MacOS/FinderSyncCollaborationFileProviderOverride (0x118979f50). One of the two will be used. Which one is undefined.
```

tatokis, 2019-09-17:
```
sudo kextcache -invalidate /
ls -la /Library/StagedExtensions/Library/Extensions/org.pqrs.driver.Karabiner.VirtualHIDDevice.v060800.kext/Contents
```

It appears macOS now copies the kexts to `/Library/StagedExtensions`, and I'll take a guess that your backup restore broke the permissions in that directory.
If the above doesn't fix it, you'll need to boot into recovery mode and either delete the kext from that directory, or fix the permissions in it.

tatokis, 2019-09-17:
Do try rebooting too, just to be sure.

neeldug, 2019-09-18:
Just rebooted and completely fixed, no idea. Thanks anyway.

closed 2019-09-18 by neeldug