ckb-next/ckb-next #461: Please add this to the various Linux repositories?

CHJ85, 2019-08-30:
Hi there. I'm a Ubuntu user. Trying to install this program.
The quickinstaller is throwing me all sorts of errors.
And besides. the terminal is kinda scary for most people.
But this is such an important program it should be easily available for people.
So could you please add it to the repos or create a Snap or a Flatpak or something?

ZirconiumX, 2019-08-30:
We intentionally avoid being in repositories where possible because our development model is fundamentally incompatible with it; we roll fixes and new device features into a new release, instead of backporting them to older "stable" releases. It results in less maintenance overhead, just like being a user-space driver instead of a kernel-space driver.

CHJ85, 2019-08-30:
Ah, I see. That makes sense.
Well. Could you at least provide a DEB package?

ZirconiumX, 2019-08-30:
The community provides those; there's a PPA for it I think. We just build from source.

CHJ85, 2019-08-30:
Ok thank you :+1:

closed 2019-08-30 by CHJ85