ckb-next/ckb-next #462: Please add support for the K50 Gaming Keyboard?

CHJ85, 2019-08-30:
Hi there.
I was wondering if you guys could please add support for the K50 Gaming Keyboard?
I know it's a few years old, but it's still a great keyboard.
And it's the one I'm using :)

ZirconiumX, 2019-08-30:
As far as I know, the Raptor doesn't actually have anything special that we can provide.

CHJ85, 2019-08-30:
What about the beautiful backlight colors?
For some reason, the colors on this model can only be controlled from their software.

tatokis, 2019-09-01:
Generate and attach a report, and also send the output of
```
sudo usbhid-dump -d 1b1c
```

psyray, 2019-12-12:
Hi, here comes the result of the command
```
003:013:002:DESCRIPTOR         1576190538.302758
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 19 00 29 6F 15 00 25 01 75 01
 95 70 81 02 C0

003:013:001:DESCRIPTOR         1576190538.303917
 05 0C 09 01 A1 01 05 0C 19 00 2A FF 0F 15 00 26
 FF 0F 75 10 95 02 81 00 C0

003:013:000:DESCRIPTOR         1576190538.304383
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 75 08 95 01 81 01 05 07 19 00
 2A FF 00 15 00 26 FF 00 75 08 95 06 81 00 05 08
 19 01 29 03 25 01 75 01 95 03 91 02 95 05 91 01
 C0
```
This keyboard have 18 macro buttons, backlight colors (16,8 millions of colors) and an internal memory of 36Ko to save game profiles and make it available across computers.

[image]

[image]

If you could add support for this keyboard, it will be great.
Thanks

gravityfargo, 2024-05-01:
This issue has been marked as stale due to a lack of activity. 

Please feel free to reopen it if you have more to add or if the issue still needs attention. However, it would be best to [open a new issue](https://github.com/ckb-next/ckb-next/issues/new/choose) and provide a newly generated report issue using our updated templates.

closed 2024-05-01 by gravityfargo