ckb-next/ckb-next #463: MacOS High Sierra support?

Japi-l, 2019-09-05:
Hey,

I noticed that the pre-built packages for Mac don't support High Sierra (10.13). I tried version 0.4.1 both the normal and legacy. And also tried the 0.4.0 (normal). But in all cases after I installed the software, and tried to open ckb-next, it just said that I need 10.14 Mojave to launch the ckb-next. So I couldnt launch the ckb-next. Has the software maybe accidentally compiled only to Mojave (10.14)?

tatokis, 2019-09-05:
Really sorry for this. My build machine is unavailable, so I asked @fleischie to build the binaries. 

We specified minimum 10.11 but apparently it didn't take effect. 
We already amended the binaries once because it was set to 10.14. Again, sorry for this. We'll try to get it sorted out asap but I don't know how long it'll take. 

You should be able to still run 0.4.0 for now. I'll let you know once we've managed to properly fix this.

fleischie, 2019-09-05:
@Japi-l Can you check again now? I uploaded the re-amended binaries.

<details>
Seemingly the correct configuration flag is not `OSX_DEPLOYMENT_TARGET`, but `CMAKE_OSX_DEPLOYMENT_TARGET`.
</details>

Japi-l, 2019-09-05:
Thanks both for the fast replies :) Now I tried it, and it works in High Sierra now!

tatokis, 2019-09-05:
Excellent. Thank you for testing and reporting the issue

closed 2019-09-05 by tatokis