ckb-next/ckb-next #464: can't open app

tarbear123, 2019-09-05:
i've installed it but I can't open the app in Linux, plz help

OvermindDL1, 2019-09-05:
How did you install it?  `./quickinstall`?

tarbear123, 2019-09-06:
with the sudo commands

OvermindDL1, 2019-09-06:
`sudo ./quickinstall`?  Shouldn't really be necessary to sudo it as I think it asks for it only when needed (and it's good to minimize that access area).

tatokis, 2019-09-06:
What do you mean by `can't open app`. Does it not start? Is there no icon to click on? Is there an error when running it? Have you tried typing `ckb-next` in a terminal? Does it not autostart on reboot?
What distro are you using? Which version?

tenoriojuann, 2019-09-09:
Same issue.

The application would not start. No error, no crashes.

I downloaded the latest release but still no luck.

I am running MacOSX 10.14

tarbear123, 2019-09-14:
> `sudo ./quickinstall`? Shouldn't really be necessary to sudo it as I think it asks for it only when needed (and it's good to minimize that access area). What are the precise commands you used and what was the output of those commands and the output of attempting to launch the program from the terminal?

the commends i am using is:
sudo apt-get install ckb-next

the output is
taran@tarans-laptop:~$ sudo apt-get install ckb-next
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package ckb-next

OvermindDL1, 2019-09-14:
`ckb-next` isn't in a PPA, hence the `E: Unable to locate package ckb-next`.

Easiest thing to do is clone this project and run quickinstall, essentially choose a permanent home for it, navigate there in the terminal, and run:
```zsh
git clone https://github.com/ckb-next/ckb-next.git
cd ckb-next
./quickinstall.sh
```

A question, where did you see that running `sudo apt-get install ckb-next` would work?  Where-ever that is documented needs to be fixed.

tarbear123, 2019-09-18:
> `ckb-next` isn't in a PPA, hence the `E: Unable to locate package ckb-next`.
> 
> Easiest thing to do is clone this project and run quickinstall, essentially choose a permanent home for it, navigate there in the terminal, and run:
> 
> ```shell
> git clone https://github.com/ckb-next/ckb-next.git
> cd ckb-next
> ./quickinstall.sh
> ```
> 
> A question, where did you see that running `sudo apt-get install ckb-next` would work? Where-ever that is documented needs to be fixed.

I put the quickinstall.sh but i get
bash: ./quickinstall.sh: No such file or directory
plz help

OvermindDL1, 2019-09-18:
> I put the quickinstall.sh but i get

What do you mean by 'put'?  Did you clone it?  Did you follow the instructions above precisely?

> bash: ./quickinstall.sh: No such file or directory

Exactly what it says, you aren't in a directory where that file exists.

tarbear123, 2019-09-18:
I did follow your instructions exactly the way you seid and I am in the correct area but I still get the error

tatokis, 2019-09-18:
Run this if you haven't already `sudo apt install build-essential cmake libudev-dev qt5-default zlib1g-dev libappindicator-dev libpulse-dev libquazip5-dev`
Then `./quickinstall`, there's no `.sh` in the end of it

@tenoriojuann Please open a new issue and generate and attach a report as instructed

tarbear123, 2019-09-18:
thanks for helping me, got it figured out

OvermindDL1, 2019-09-18:
> Then ./quickinstall, there's no .sh at the end of it

Oh whoops, forget that, force of habit...  ^.^;

closed 2019-09-18 by tatokis