ckb-next/ckb-next #469: Share state of modifier keys with all connected keyboards

a-chaudhari, 2019-09-13:
Mouse: Scimitar Pro
Mac 10.13 High Sierra

From my understanding ckb uses the karabiner kernel module under the hood.  One useful behavior of karabiner is the modifier keys are shared between all keyboards.  Meaning I can press "command" on one keyboard and "tab" on the other and use that to cycle through open programs.  Without this sharing I'd be sending separate "command" and "tab" keys instead of a single "command+tab".

This behavior is particularly handy when playing MMOs as this allows one to use modifier keys on the keyboard while pressing mouse buttons to have easy access to more than one set of spells.

Unfortunately ckb and karabiner don't work at all together. And ckb doesn't share modifier keys.  Is there anyway to extend the behavior of ckb to bring this feature from karabiner over?

tatokis, 2019-09-17:
Yeah, unfortunately modifiers aren't shared, as it should be a feature in the OS, not the input driver.

It's something to consider in the future, but at the moment it's not high in the priority list as it can be quite messy to implement.

For the time being, you'll have to make macros for each combination you want, as with a macro you can press the modifier using the emulated mouse device.
For example `+lshift,+l,-l,-lshift`

Alternatively can bind one of the mouse buttons to shift itself.

tatokis, 2019-09-21:
Can you please try this build? There should be an option under settings to enable shared modifiers.
I am currently unable to test it, so if something goes wrong, please downgrade to the latest release.

http://ckb-next.info/mac/ckb-next_v0.4.1-22-ge10417a_10.12-SDK.dmg

```
0c83cb9dc69fece5bd9f942ad2bb824187e03e236b0786ba6432a89197ad4305  ckb-next_v0.4.1-22-ge10417a_10.12-SDK.dmg
```

a-chaudhari, 2019-09-24:
Hello, it has no effect.  Modifiers are still not shared.

tatokis, 2019-10-06:
Should be fixed now

http://ckb-next.info/mac/ckb-next_v0.4.1-24-g6024721_10.12-SDK.dmg

```
dadbcc8954bc1c159e42bf80afaa2396e568802f135a7b5ea84d1cdae676ef83  /home/tatokis/ckb-next/ckb-next_v0.4.1-24-g6024721_10.12-SDK.dmg
```

a-chaudhari, 2019-11-02:
Sorry for the delay.  It doesn't work for me

tatokis, 2019-11-09:
I am going to guess you do not have a Corsair keyboard, then.
If that's the case, then there is not much that can be done.

Unfortunately there's no reliable way to receive keystrokes from other devices the way ckb is designed and how Apple changed their APIs. It barely functions as-is on Catalina.

There are most likely other applications that work at a higher level that can perform such operations using the accessibility APIs to work around this issue.

a-chaudhari, 2019-12-07:
Thanks for the update. Correct, I do not have a corsair.  it's a Happy Hacker Keyboard.  

Karabiner has support for all keyboards, do you know how they implement it?

closed 2022-10-03 by tatokis