ckb-next/ckb-next #472: Harpoon on Ubuntu 19.04 Couldn't open device

markfid, 2019-09-20:
I have installed the CKB-Next software on Ubuntu 19.04 and the Corsair Harpoon doesn't seem to be found. Here are the device details form under the mouse:

Corsair Harpoon RGB
Model: RGP0079

Here is the detect report:

ckb-next-dev-detect - Fri Sep 20 12:39:20 BST 2019

Kernel: Linux XPS-13 5.0.0-21-generic #22-Ubuntu SMP Tue Jul 2 13:27:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

OS Version:
Distributor ID:	Ubuntu
Description:	Ubuntu 19.04
Release:	19.04
Codename:	disco

USB Devices:
Couldn't open device, some information will be missing

Bus 001 Device 005: ID 1b1c:1b5e Corsair 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1b1c Corsair
  idProduct          0x1b5e 
  bcdDevice            0.06
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x005b
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      60
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      27
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x04  EP 4 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      21
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1

Dmesg on current boot:
Sep 20 12:35:15.340175 XPS-13 kernel: usb 1-1: New USB device found, idVendor=1b1c, idProduct=1b5e, bcdDevice= 0.06
Sep 20 12:35:15.340665 XPS-13 kernel: usb 1-1: Product: CORSAIR HARPOON RGB WIRELESS Gaming Mouse
Sep 20 12:35:15.340838 XPS-13 kernel: usb 1-1: Manufacturer: Corsair
Sep 20 12:35:15.404115 XPS-13 kernel: input: Corsair CORSAIR HARPOON RGB WIRELESS Gaming Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1B1C:1B5E.0003/input/input33
Sep 20 12:35:15.464069 XPS-13 kernel: hid-generic 0003:1B1C:1B5E.0003: input,hidraw2: USB HID v1.11 Mouse [Corsair CORSAIR HARPOON RGB WIRELESS Gaming Mouse] on usb-0000:00:14.0-1/input0
Sep 20 12:35:15.464506 XPS-13 kernel: hid-generic 0003:1B1C:1B5E.0004: hiddev0,hidraw3: USB HID v1.11 Device [Corsair CORSAIR HARPOON RGB WIRELESS Gaming Mouse] on usb-0000:00:14.0-1/input1
Sep 20 12:35:15.464805 XPS-13 kernel: hid-generic 0003:1B1C:1B5E.0005: hiddev1,hidraw4: USB HID v1.11 Device [Corsair CORSAIR HARPOON RGB WIRELESS Gaming Mouse] on usb-0000:00:14.0-1/input2

Dmesg on last boot:

Daemon on current boot:
-- Logs begin at Tue 2019-04-23 08:29:33 BST, end at Fri 2019-09-20 13:32:32 BST. --
Sep 20 13:32:03.139399 XPS-13 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Sep 20 13:32:03.151680 XPS-13 ckb-next-daemon[1000]: ckb-next: Corsair RGB driver 0.4.0
Sep 20 13:32:03.152063 XPS-13 ckb-next-daemon[1000]: [I] Root controller ready at /dev/input/ckb0
Sep 20 12:34:32.254068 XPS-13 ckb-next-daemon[1000]: [I] Caught signal 15
Sep 20 12:34:32.254068 XPS-13 ckb-next-daemon[1000]: [I] Closing root controller
Sep 20 12:34:32.254446 XPS-13 ckb-next-daemon[1000]: [I] Removed device path /dev/input/ckb0
Sep 20 12:34:32.254105 XPS-13 systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Sep 20 12:34:32.254389 XPS-13 systemd[1]: ckb-next-daemon.service: Succeeded.
Sep 20 12:34:32.254904 XPS-13 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.
Sep 20 12:34:32.891106 XPS-13 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Sep 20 12:34:32.892550 XPS-13 ckb-next-daemon[7380]: ckb-next: Corsair RGB driver 0.4.1
Sep 20 12:34:32.892842 XPS-13 ckb-next-daemon[7380]: [I] Root controller ready at /dev/input/ckb0

Daemon on last boot:
-- Logs begin at Tue 2019-04-23 08:29:33 BST, end at Fri 2019-09-20 13:32:32 BST. --
Aug 06 18:37:27.988722 XPS-13 systemd[1]: Started Corsair Keyboards and Mice Daemon.
Aug 06 18:37:28.031540 XPS-13 ckb-next-daemon[1083]: ckb-next: Corsair RGB driver 0.4.0
Aug 06 18:37:28.031540 XPS-13 ckb-next-daemon[1083]: [I] Root controller ready at /dev/input/ckb0
Aug 06 18:37:36.669419 XPS-13 ckb-next-daemon[1083]: [I] Caught signal 15
Aug 06 18:37:36.669419 XPS-13 ckb-next-daemon[1083]: [I] Closing root controller
Aug 06 18:37:36.669419 XPS-13 ckb-next-daemon[1083]: [I] Removed device path /dev/input/ckb0
Aug 06 18:37:36.669055 XPS-13 systemd[1]: Stopping Corsair Keyboards and Mice Daemon...
Aug 06 18:37:36.687479 XPS-13 systemd[1]: ckb-next-daemon.service: Succeeded.
Aug 06 18:37:36.688030 XPS-13 systemd[1]: Stopped Corsair Keyboards and Mice Daemon.

Dev nodes:
/dev/input/ckb0
/dev/input/ckb0/pid
7380
/dev/input/ckb0/version
0.4.1
/dev/input/ckb0/connected

Environment:
QT_ACCESSIBILITY=1
QT4_IM_MODULE=xim
XDG_SESSION_DESKTOP=ubuntu
XDG_CURRENT_DESKTOP=ubuntu:GNOME
DISPLAY=:0
QT_IM_MODULE=ibus

Kernel command line:
BOOT_IMAGE=/boot/vmlinuz-5.0.0-21-generic root=UUID=c10dae53-1615-4b64-a03e-0add3faf71ce ro quiet splash vt.handoff=1

tatokis, 2019-09-20:
The wireless version of this mouse isn't supported and won't be for some time. There's an issue open about it already.

markfid, 2019-09-20:
Ohh, I'm actually using it wired via the cable, but guess its still the same issue. Can you let me know which the the bug ID for it so that I can track it please?

tatokis, 2019-09-20:
Here you go https://github.com/ckb-next/ckb-next/issues/346

closed 2019-09-20 by tatokis

markfid, 2019-09-20:
Cheers