ckb-next/ckb-next #473: Question: likelihood of support for non-Corsair devices?

cgku, 2019-09-20:
I recently moved from Corsair to get an RGB tenkeyless with brown switches; no shock that this vendor doesn't offer linux support.  Having not found any other tool to support RGB keyboards on Linux, I was curious what is the likelihood that ckb-next will/can support other vendor's products (or something generic) in the near future.  My guess is this program is tailored for a Corsair protocol and not some generic set of instructions.  I don't have enough background in this topic to look at the code myself, so I must depend upon the know-how of others to tell me.

Thank you in advance, and nice work on this utility!

tatokis, 2019-09-29:
I am willing to accept PRs for other devices as long as there is no other driver for them, and the code is clean.
However, I will not work on them directly. I don't even have enough time and mental strength to keep Corsair devices supported.

OvermindDL1, 2019-09-30:
> I am willing to accept PRs for other devices as long as there is no other driver for them, and the code is clean.

Hmm, so just to clarify, code to add support for the old Phillips amBX system would be accepted?  It has lights, fans, vibration bar, and a couple key mappings (used to toggle some settings and adjust volume and so forth).  I'm using a custom-made hacked-together Haskell program as I couldn't find anything else useful out and just wanted something working (almost 10 years ago now I think), and would love for it to fit into the rest of my corsair things.  The lights would be easy to map, the animations would fit perfectly on them.  Fan's should be able to be put in somehow, simple 0% to 100% power for them is all they take (plus patterning for adjusting flow rapidly).  The vibration bar is more complicated so probably leave it out (I don't use it anyway except to surprise people that use my computer to be honest) though at least exposing a mapping to send commands to it via the usual other things would be convenient, no real gui work needed as not much is useful for it...

The Phillips amBX USB protocol is dead simple, though entirely undocumented (figured it out manually, was a bit of a pain, but I've got everything I care about it figured out).  Everything else online for linux for it is essentially some example python code and so forth but nothing useful.

cgku, 2019-10-01:
@tatokis My keyboard is a Tecware Phantom, an amazing little keyboard considering the price point.  The vendor only provides Windows drivers/software.  I have no idea how clean their code is, but if it happens to get accepted for support, that would be amazing.  If no, I completely understand.

RattyDAVE, 2019-11-23:
Suggestion : ASRock Motherboard 

I have done some initial scripting and have managed to extract the core commands and control them using i2cset.

Examples are here : https://github.com/RattyDAVE/asrock-leds/blob/master/changeled.sh

DAVE

JoshDreamland, 2019-12-06:
I looked into adding support for ASUS keyboards, but the Corsair protocol seems to be included in various high-level functions directly. The entire backend is C. There's nowhere to shove a vtable between the UI and the driver to translate high-level concepts (set key color, rotate hardware profile) into low-level USB messages.

I would love to add support for ROG devices if a developer could either install a system for registering new protocols, or give me the go-ahead to wrap the corsair logic in a struct of some kind.

closed 2022-05-29 by Ravenslofty