ckb-next/ckb-next #477: Macro Delay not working

DLCIncluded, 2019-09-29:
## General information
Linux Mint 19.2
Corsair K55

Attempting to create a macro with delays between keypresses, the delay does not matter, 100 is the same as 100000. I have even tried using one of the examples as a base:
+d,-d,+e=5000,-e,+l,-l=10000,+a,-a,+y,-y=1000000,+enter,-enter
and it immediately types "delay" and a return in the blink of an eye as if you pasted the text in. 

tried recording as well as manually typing the macros. 

Everything else, ie rgb, animations, etc seem to work beautifully.

### Source
Tried both installing from tatokis/ckb-next repo as well as building using the quickinstall script

## Logs & crash reports
Not sure where logs are held if they're needed please let me know where they are and will happily supply them.

DLCIncluded, 2019-10-05:
I have confirmed with a K95 as well, and the issue persists. 
Both my K95 and K55 work with the macro delay in windows. 
Its not the end of the world that this doesnt work, just an inconvenience. 
If someone gets time to look into it, that would be lovely.

tatokis, 2019-10-05:
Can you perhaps try downgrading to 0.4.0? I tried macros on master recently and I couldn't get it working properly either. 

I'm not sure what happened

DLCIncluded, 2019-10-05:
Hey tatokis, thanks for replying. 
I just tried purging all of CKB, then building and installing 0.4.0 and still getting the same. 
I tried recording a macro, pressed one, count to 3, press two count to 3, press three and it created:
+1=77692,-1=2405176,+2=94033,-2=2226147,+3=82002,-3
which just types 123 right away ignoring my delay. 

I noticed there is a "Generate Report" option so I did that and attached the result:
[ckb-next-report-1570283887.gz](https://github.com/ckb-next/ckb-next/files/3693304/ckb-next-report-1570283887.gz)
screenshot of macro
[image]
screenshot of version
[image]

Thanks for your help and looking into this

DLCIncluded, 2019-10-05:
I just went to 0.3.2 and its still not working for me. I'm starting to wonder if its something on my end.

robospectrum, 2020-11-06:
the same for me. Any news?

tatokis, 2020-11-06:
Are you pressing apply after recording the macro?
Can you try the latest `ckb-next-git` package/code from git?

closed 2022-10-03 by tatokis