ckb-next/ckb-next #478: MacOS Catalina - k68 - keyboard unresponsive after reboot on lock screen

justinhauer, 2019-10-03:
## General information

### Source
How and where did you get this program, e.g.:
Installed via master branch

## Logs & crash reports

Attach:
 * The generated report from the section above
 * Any additional crash reports you've managed to gather

## What is the program's current behaviour?
Upon restarting OS, the keyboard is unresponsive at the login prompt. A second keyboard is needed to log in. After logging in, CKB remembers profiles and the keyboard functions normally.
## What is the expected behaviour of the program?
I can use my corsair keyboard to log into my Mac
## Any other notes & comments?
I am on the latest public beta for Catalina - I think Catalina is close to the gold image so it is close to the GA of what will be put out for all macs.

tatokis, 2019-10-03:
Please generate and attach a report after the issue has occurred.

fleischie, 2019-10-04:
To be perfectly honest: I never had any external (non-Apple) keyboard working on the macOS login screen.

Is this only an issue with corsair keyboards?

ItsEcholot, 2019-10-05:
My k70 works perfectly on the login screen so I can't confirm this. Also on latest public Catalina Beta build.

justinhauer, 2019-10-06:
I disabled filevault and the keyboard works at the login screen after startup/reboot. I'm guessing there is as sort of issue loading the drivers in the user profile otherwise? I'd rather keep filevault but it seems like the only way to remedy the situation

jclapp, 2019-10-31:
Hi Justin, not sure if this will help you, but it helped me on macOS Catalina - and I am also using FileVault.  I am using a K68 and it seemed like only a strange selection of keys worked to enter the password after reboot.  There is a key sequence to toggle between BIOS compatibility mode and normal mode.  Hold down the F1 key and Lock key (the button to the right of the brightness button) for about 5 seconds.  This toggles modes and allows me to enter the password at bootup.  Switching out of BIOS compatibility mode re-enables the multimedia keys and other features.  So kind of a pain, but not too bad.