ckb-next/ckb-next #480: Super Key/Windows Key does not respond on MacOS (Strafe RGB)

ClaudiusCluever, 2019-10-07:
## General information
I am on MacOS 10.14.6 with a Strafe RGB

### Source
  * Pre-built dmg for macOS

## What is the program's current behaviour?
When ckb-next is running and shown in the top bar, the windows key (super key, normally mapped as cmd on mac) does not respond. When I quit ckb-next all is working as usual.
Since this is so, it doesn't break functionality for me. I am primarily filing this to help catch bugs early that may break things for other people.

## What is the expected behaviour of the program?
The windows key working as command on MacOS with ckb-next running.

tatokis, 2019-10-07:
You most likely have winlock enabled. Press the winlock button on your keyboard and it should work again, assuming you haven't bound that button to something else.

ClaudiusCluever, 2019-10-08:
Right, that did it. Maybe it would be smart to build in a Notification about this?

brallanAguilar, 2021-03-11:
> You most likely have winlock enabled. Press the winlock button on your keyboard and it should work again, assuming you haven't bound that button to something else.

Worked for Linux too, thanks! I suggest to give feedback to the user

closed 2022-06-27 by tatokis